Makuna/NeoPixelBus #1: Change pixel number at run time

sticilface, 2015-05-06:
Would be useful to be able to redefine the number of pixels, and the output pin.  

The esp can be configured by web interface, so modules can be deployed then configured. 

I tried adjusting the count by re-issuing the NeoPixelBus strip = NeoPixelBus(pixelCount, 8); (in the adafruit lib, not yours) but every time the loop circles it gets reset.

Makuna, 2015-05-06:
The best way to handle this today is to dynamically create the NeoPixelBus rather than doing it statically.  See below.

You would not be able to maintain pixel colors and not sure you would want to.

```
NeoPixelBus* pNeoPixels = NULL; // dynamic

setup()
{
    uint16_t countPixels = Eeprom.GetPixelCount(); // how ever you store them
    uint8_t pinPixels = Eeprom.GetPixelPin(); // how ever you store them
    ChangeNeoPixels(countPixels, pinPixels); // initial setup
}

loop()
{
...
}

// called to change the number of pixels
void ChangeNeoPixels(uint16_t count, uint8_t pin)
{
    if (pNeoPixels)
    {
        delete pNeoPixels;
    }
    pNeoPixels = new NeoPixelBus(count, pin);
}
```

sticilface, 2015-05-06:
Thank you. 

Is that not able to maintain them when the number has been changed. But after it works as normal?

Makuna, 2015-05-06:
What does it mean when you change the number of pixels?
To me, this means its a completely different physical collection of LEDs, so maintaining the colors that are present doesn't logically make sense.  What example do you have that you would want to maintain the colors?

In the example above, it will clear the internal state of the "new" collection of LEDs to black.  Just call Show() and it will be applied to the physical LEDs.

sticilface, 2015-05-06:
We're talking about the same thing. And I agree you don't want that anyway. So we're in agreement. 

I wanted to make sure that it didn't affect the function of the pixels. 

Thanks

A

sticilface, 2015-05-08:
Worked very well.  Thank you

closed 2015-05-08 by sticilface

mariusmotea, 2019-02-08:
Hi,

I tried this option with the current version of the library and WS2812 board, but is not working.

```

NeoPixelBus<NeoGrbFeature, Neo800KbpsMethod> strip(false);

void ChangeNeoPixels(uint16_t count)
{
    if (strip)
    {
        delete strip;
    }
    strip = new NeoPixelBus(count);
    strip->Begin();
}
```

the error is:

```
WS2812BHueStrip:577:17: error: expected type-specifier before 'NeoPixelBus'
     strip = new NeoPixelBus(count);
                 ^
WS2812BHueStrip:577:17: error: expected ';' before 'NeoPixelBus'
WS2812BHueStrip:578:10: error: base operand of '->' has non-pointer type 'NeoPixelBus<NeoGrbFeature, NeoEsp8266DmaMethodBase<NeoEsp8266DmaSpeedWs2812x> >'
     strip->Begin();
          ^
exit status 1
could not convert 'strip' from 'NeoPixelBus<NeoGrbFeature, NeoEsp8266DmaMethodBase<NeoEsp8266DmaSpeedWs2812x> >' to 'bool'
```

I try to create a ws2812 sketch that store configuration in SPIFS so the number of leds can be changed from web interface.

Thanks,
Marius.

Makuna, 2019-02-08:
@mariusmotea commenting on a closed issue is not a good practice.  

Please read this [Wiki Topic on the subject of dynamic pixel counts](https://github.com/Makuna/NeoPixelBus/wiki/FAQ-%233).

mariusmotea, 2019-02-08:
Thanks, i found that but it copile with errors.  After i replace 'Strip' with "strip' and 'strip.SetPixelColor' with 'strip->SetPixelColor` it compile but i did not test it to confirm.