Makuna/NeoPixelBus #2: Working with the RGB object

sticilface, 2015-05-08:
I'm trying to get certain bits of the adafruit effects to work with your lib, and I'm struggling. 
The adafruit guys have wheel function that allows generation of smooth colours, but it usually returns a 32bit number for the colours.  your lib works with r,g,b object.

now you have this in you rgb.cpp which suggests to me that you can return an rgb object with 3 colours from a function... however...

'''''
RgbColor RgbColor::LinearBlend(RgbColor left, RgbColor right, uint8_t progress)
{
	return RgbColor( left.R + ((right.R - left.R) * progress / 255),
		left.G + ((right.G - left.G) * progress / 255),
		left.B + ((right.B - left.B) * progress / 255));
}
''''''

This does not work for me.

`````
RgbColor Wheelnew (byte WheelPos) {
  WheelPos = 255 - WheelPos;
  if(WheelPos < 85) {
   return  RgbColor(255 - WheelPos * 3, 0, WheelPos * 3);
  } else if(WheelPos < 170) {
   WheelPos -= 85;    
   return RgbColor(0, WheelPos * 3, 255 - WheelPos * 3);
  } else {
   WheelPos -= 170;
   return  RgbColor(WheelPos * 3, 255 - WheelPos * 3, 0);
  }
}  
```````
I get 
Melvano-Core.ino:36:1: error: 'RgbColor' does not name a type
Error compiling.

Makuna, 2015-05-08:
I cut and pasted your snippit of code and it compiles without error for me.

Make sure that that NeoPixelBus.h was included before this function.

```
#include <NeoPixelBus.h>
```

If you are still having an issue, include the complete INO and I will confirm.

sticilface, 2015-05-08:
right, it would seem like that is also the case for me.  

It works in your example.
It working in your example with the dynamic additions
does not work in my main project. 
The code for this project is a mess, it everything I've been working on in one.  And i've only been doing C for a number of weeks, so you're probably better off not going anywhere near it.  having said that I'm quite confused, it is the first time i've got an error like this, without it giving me the actual location in the code. it just fails, saying a random place in the first into file, which contains all my headers. 

i've got the right includes so its not that.  and the RgbColor object works fine when just being called normally!

`````
#include <NeoPixelBus.h>
#include <RgbColor.h>
``````

any ideas of what to try?

Makuna, 2015-05-08:
I suspect environment variable problem.  Something is on the path that shouldn't be.  The setup of the environment from the main github is nasty and error prone.
 
This is going to sound draconian. Remove your complete Environment (but keep your source code) and start over building the environment up.  This includes removing any environment variables you added.

When building it back, try this simpler set of steps.
1) Install the latest official Arduino Ide from Arduino.cc
2) Enlist into https://github.com/sandeepmistry/esp8266-Arduino
3) Copy esp8266com directory from this enlistment into your documents\arudino\hardware\ folder (created the hardware folder if not present).

Then start the Arduino IDE, make sure to select the esp8266 as a target, load your ino, and build.

sticilface, 2015-05-08:
Oh, for some reason my second message did not come up!  weird...

I found the problem.  It is to do with the orders of #includes as well as, if i declare a variable before the neopixelbus include.  
If this....
``````
extern "C" {
#include "ets_sys.h"
#include "c_types.h"
#include "osapi.h"
#include "user_interface.h"
}
``````
is before this
```````
#include <NeoPixelBus.h>
#include <RgbColor.h>
```````

it fails to build *** BUT ONLY if i try to include a function that returns the rgbcolor object!!!!  

it also fails if i declare a variable before it as well.

question... is this normal behaviour...?
Thank you for your help to, as this is not really related to your lib.

Makuna, 2015-05-08:
You should not directly include any of those specific system level headers like osapi.h.  Replace that with just #include <Arduino.h> and it should then work.

I ran into something similar that works fine when I build a library for AVR Arduino that fails for esp8266 Arduino, by replacing my includes with just this one it fixed the issues.

Sent from Windows Mail

From: sticilface
Sent: ‎Friday‎, ‎May‎ ‎8‎, ‎2015 ‎10‎:‎33‎ ‎AM
To: Makuna/NeoPixelBus
Cc: Makuna@hotmail.com

Oh, for some reason my second message did not come up! weird...

I found the problem. It is to do with the orders of #includes as well as, if i declare a variable before the neopixelbus include.

If this....
extern "C" {
#include "ets_sys.h"
#include "c_types.h"
#include "osapi.h"
#include "user_interface.h"
}

is before this
#include <NeoPixelBus.h>
#include <RgbColor.h>

it fails to build *** BUT ONLY if i try to include a function that returns the rgbcolor object!!!! 

it also fails if i declare a variable before it as well.

question... is this normal behaviour...?
Thank you for your help to, as this is not really related to your lib.

Makuna, 2015-05-11:
Did this fix the issue for you?

sticilface, 2015-05-11:
I will have to get back to you, as I'm having a totally nightmare at the moment trying to get to grips with git and source tree!!!!  

Basically I've made some code that handles wifi, mqtt, debugging info, web server etc for my esp. 
This code i want to keep common to a bunch of projects that do different things. I'm not sure how well git will do this.  but with zero experience.... I'm learning the hard way. 

anyway.  this is my first couple of months really doing any coding... and guess what... i've lost a bunch of changes... can't merge back to my master files....  indeed up with so much conflict stuff that i can't compile my code...  its all a bit frustrating... 

changing the order of things helped though... 
I needed those header files for the original ws2812 stuff that i made, and i just didn't clear them out!

Makuna, 2015-05-11:
We are way off topic, but oh well ;-)  Consider using # if to define sections for each.

If you look at the NeoPixelBus.cpp, you see it has compile branching statements like...
```
#if defined(ESP8266)

#else

#endif
```
This allows for you to share the same code for different systems.  This one file can be compiled for some five or six different versions of the same code within based on the hardware platform.

sticilface, 2015-05-11:
It is of topic….  but thanks… 

So here is the odd thing… I’ve managed to revet back to something that was not screwed up.  
I’ve taken out the references to c extern…  they were actually there from ages ago before the ESP.h came about…  
I’ve included the neopixelbus header and my code, and it compiles just fine… 

now if i use an 
```````#ifdef xyz
my code
#endif```````````

I get that error again when it is not defined, so something is not quite right.  it should be just missing it out . 

A
> On 11 May 2015, at 19:09, Michael Miller <notifications@github.com> wrote:
> 
> We are way off topic, but oh well ;-) Consider using # if to define sections for each.
> 
> If you look at the NeoPixelBus.cpp, you see it has compile branching statements like...
> 
> #if defined(ESP8266)
> 
> #else
> 
> #endif
> This allows for you to share the same code for different systems. This one file can be compiled for some five or six different versions of the same code within based on the hardware platform.
> 
> —
> Reply to this email directly or view it on GitHub <https://github.com/Makuna/NeoPixelBus/issues/2#issuecomment-101002253>.
>

sticilface, 2015-05-11:
i suspect now that this is what I’m finding

https://code.google.com/p/arduino/issues/detail?id=906

Makuna, 2015-05-12:
That only effects the INO file.

Moved shared code into a .h and .cpp and they should work there.

Sent from Windows Mail

From: sticilface
Sent: ‎Monday‎, ‎May‎ ‎11‎, ‎2015 ‎2‎:‎23‎ ‎PM
To: Makuna/NeoPixelBus
Cc: Makuna@hotmail.com

i suspect now that this is what I’m finding

https://code.google.com/p/arduino/issues/detail?id=906

closed 2015-05-20 by Makuna