Makuna/NeoPixelBus #4: Dimming

sticilface, 2015-05-15:
(Forgive me as I have a few issues)... 

I can't seem to get the dimming to work reliably...  I have a slider on a webpage, 0-255.. which returns a string...  

```````
void WS2812_dim_string (String Value)
{

  RgbColor prevColor;

      int dim = Value.toInt();
      if (dim > 255) dim = 255;
      if (dim < 0) dim = 0;

      int DiffDim = CurrentBrightness - dim;
      CurrentBrightness  = dim; 

      if (DiffDim > 0) {  
        Serial.println("Brightness Darken by: " + String(DiffDim));

        for (int i = 0; i < pixelCount; i++) {
          prevColor = strip.GetPixelColor(i);
          prevColor.Darken(DiffDim);
          strip.SetPixelColor(i,prevColor);
        }
      } else if (DiffDim < 0) { 
        DiffDim = DiffDim * -1;
        Serial.println("Brightness Lighten by: " + String(DiffDim));
          for (int i = 0; i < pixelCount; i++) {
          prevColor = strip.GetPixelColor(i);
          prevColor.Lighten(DiffDim);
          strip.SetPixelColor(i,prevColor);       
        }
      } 
          strip.Show();
}
`````````

Making it darker works just fine, but going the other way results in far too much white most of the time.  Got any ideas?  If you darken by a negative does that make it lighter? should it...?

Makuna, 2015-05-19:
RgbColor.Darken will blend the color toward black, and RgbColor.Lighten will blend the color toward white.  It does not retain the original color so if you Darken or Lighten is successive calls, each time it will start with the color you already have as it knows nothing about the original color.  

The parameter is unsigned, so negative has no meaning to the methods. 

If the "effect" you are trying to build is to Darken from a Specific color toward black and then back to the original color, then the correct way to do this is to use the LinearBlend method; blend from the "Original" color to Black, then blend of the current color back toward the "original" color.

Makuna, 2015-05-20:
the other alternative is to track the dim level, and always apply it to the original color.

```
RgbColor primeColor(192,12,60);

      int dim = Value.toInt();
 
       if (dim > 0) {
            // brighten
            primeColor.Lighten(max(dim, 255));
        }
        else {
            // negative means darken
            primeColor.Darken(max(abs(dim), 255));
        }
        for (int i = 0; i < pixelCount; i++) {
          strip.SetPixelColor(i,primeColor);
        }

```

sticilface, 2015-05-20:
Cheers.  I think what i am going to do, is add the dim state to every call to set a colour...  if i want the strip to be (200,0,0) then i take that and scale is with the dim state... every update will then affect it..  this way what the current colour is becomes irrelevant.. for animations... i just darken by the dim amount before applying every pixel..  

currently though... my whole set up has become a crashing mess!  not sure why.. but it just reboots constantly!

closed 2015-05-25 by sticilface