Makuna/NeoPixelBus #6: can't compile in ardu ide

matroll, 2015-05-19:
hey makuna, thanks for the work on neopixelbus.
trying to use it for the esp but can't compile in the ardu ide (1.6.4)
i get a bunch of "not defined in this scope" (below)

i followed your instructions, also tried putting the lib in <sketch path>/libraries (as recommended in new ide version) and also did an official lib install via zip import (also in 1.6.4) and reastarted the ide with each try. to no avail. thanks for your support!

C:\xxx\libraries\NeoPixelBus\NeoPixelBus.cpp: In function 'void send_pixels_800(uint8_t*, uint8_t*, uint8_t)':
C:\xxx\libraries\NeoPixelBus\NeoPixelBus.cpp:88:41: error: '_BV' was not declared in this scope
     const uint32_t pinRegister = _BV(pin);
                                         ^
C:\xxx\libraries\NeoPixelBus\NeoPixelBus.cpp:93:23: error: 'class EspClass' has no member named 'getCycleCount'
     cyclesStart = ESP.getCycleCount() + CYCLES_800;
                       ^
C:\xxx\libraries\NeoPixelBus\NeoPixelBus.cpp:109:35: error: 'class EspClass' has no member named 'getCycleCount'

Makuna, 2015-05-19:
If you have 1.6.4, please make sure you install the esp8266/Arduino with the new board manager install (copied instructions below).  BUT, remove any old esp8266/Arduino installs first! 

Then open one of the NeoPixelBus samples from the IDE samples menu and build.  Do you get the same issue?

Installing with Boards Manager
Starting with 1.6.4, Arduino allows installation of third-party platform packages using Boards Manager. We have packages available for Windows, Mac OS, and Linux (32 and 64 bit).
•Install Arduino 1.6.4 from the Arduino website.
•Start Arduino and open Perferences window.
•Enter  http://arduino.esp8266.com/package_esp8266com_index.json  into Additional Board Manager URLs field. You can add multiple URLs, separating them with commas.
•Open Boards Manager from Tools > Board menu and install esp8266 platform (and don't forget to select your ESP8266 board from Tools > Board menu after installation).

matroll, 2015-05-21:
AWESOOOOOOME!
thanks for the prompt answer makuna (was outta town until today). that worked! i had the adafruit json which was the only one available at the time. your link to the new json got it to work.
thanks again for the great work on this lib. now i'll get some info off the net to display on my pixels ;))

closed 2015-05-21 by Makuna