Makuna/NeoPixelBus #8: esp8266 example

achingbrain, 2015-05-23:
The readme mentions esp8266 support - do you have any code examples?  I'm trying to get it working with the [esp8266 Arduino library](https://github.com/esp8266/Arduino) and a couple of NeoPixels attached to GPIO2.  My sketch looks like this:

```c
#include <NeoPixelBus.h>

#define PIN 4
#define PIXELS 2

NeoPixelBus strip = NeoPixelBus(PIXELS, PIN);

void setup(void) {
  Serial.begin(115200);
  
  strip.Begin();
  
  strip.SetPixelColor(0, RgbColor(255, 0, 0));
  strip.SetPixelColor(1, RgbColor(255, 0, 0));
  
  strip.Show();
}

void loop(void) {
}
```

I'd expect them to both turn red but instead they just show random colours.

achingbrain, 2015-05-24:
Looks like I had the pin wrong in the code above - GPIO2 should be pin 2, shock horror.

Weirdly the green LED in the first NeoPixel appears to be on all the time reguardless of what value I pass into `SetPixelColor`, maybe it's fried. The second one works as expected.

achingbrain, 2015-05-24:
Actually, just tested the lights with an Arduino and they both work fine.

Makuna, 2015-05-24:
Which "type" of NeoPixels are you using? 

One thing I found was that "most" versions of NeoPixel or like LEDs require that the IO pin voltage to be within +-0.5v of the voltage powering the LEDs.  Further, "most" also require a minimum voltage of 3.5v.  The effect was similar to what you are seeing if you power the LEDs with 5v and use the esp8266 3.3v IO.  I power my pixels with 3.6v.

achingbrain, 2015-05-27:
They are [WS2812s](http://proto-pic.co.uk/led-rgb-strip-addressable-bare-5m/).  I haven't tried 3.6v but I see the green LED stuck on at about half power when running either 3.3v or 5v through both the esp and the LEDs and off with an Arduino and the Adafruit NeoPixel library at the same voltages.

Interestingly if I'm connected to the esp via the Arduino IDE serial monitor and I see the esp reset itself, sometimes the green LED does what it's told and turns itself off.

Makuna, 2015-05-28:
Please pick up the latest NeoPixelBus library and try again.  A "fix" was included to make the first pixel get updated more consistently.  Let me know if it doesn't work for you.

Makuna, 2015-05-28:
https://github.com/Makuna/NeoPixelBus/commit/503c7e3d7b3eeea11dd803efcab0616596738296

closed 2015-05-28 by Makuna

achingbrain, 2015-05-28:
I won't have access to any kit for a few days, sorry. Will test this as soon as I can. Thanks for looking in to it.

achingbrain, 2015-06-15:
Looks like you've captured the green pixel issue in #11 - I've been tracking your progress in Gitter and via that issue.  The fix in 503c7e3 didn't work for me but I rebuilt with master this morning and the green pixel problem doesn't seem to have gone away.  Nice work!