Makuna/NeoPixelBus #9: Fast allocation of pixels

sticilface, 2015-05-25:
I'm currently reading pixel colors from a buffer, or buffer array. 

Is there a really fast way to shift the data out to the array that show() uses. Right now I for loop through, generate an rgbcolor object and setpixelcolor but I figure there is a more efficient way as they are just strings of bytes? 

(Thanks for all the help btw!)

Makuna, 2015-05-26:
There is a way to get the buffer that is used, it is exposed as NeoPixelBus.Pixels().  You can write directly into this buffer.  The buffer is 3 * NeoPixelbus.PixelCount is size.

The issue is that this buffer is what is directly spewed out to the NeoPixels unmodified.  So the order of the bytes depends on what flags you defined when you created the NeoPixelsBus.  By the default, it is Green, Red, Blue, then the next pixels.  See flags NEO_GRB, NEO_RGB, and NEO_BRG for all the currently supported options.

Further, RgbColor does not just map to this, its bytes are always R, G, B.

G1, R1, B1, G2, R2, B2, G3, R3, B3, ...

if you defined a struct like this, then cast the Pixels to this, you can just iterate them.
```
struct GrbPixel
{
    uint8_t G;
    uint8_t R;
    uint8_t B;
};

...
   GrbPixel* pixels = (GrbPixels*)NeoPixelBus.Pixels();

  for (int I = 0; I < NeoPixelBus.PixelCount; I++) {
    if (pixels[I].R == 255) {
        pixels[I].R = 0;
    }
    else {
        pixels[I].R++;
    }
  }
```

sticilface, 2015-05-27:
Thank you!  However, in that one example you've introduced two concepts that I'm not familiar with at all really.  well maybe even 3. 

1) struct
2) casting
3) (pointers.. although i know a bit about them)

Your code takes an RGB object and casts it to the buffer directly using a pointer? (is that right)?  

Now.. if i have a buffer already... that is filled with G1, R1, B1, G2, R2, B2, G3, R3, B3, ...
say for example a UDP packet... or a buffer filled from serial...  can I just do something like

```
 for (int I = 0; I < NeoPixelBus.PixelCount; I++) {
pixelbuf[I] = [insert something here];
}
```

or do you have to define a struct, and cast to it...  ?

sticilface, 2015-05-27:
Actually... forget that.  My bytes are RGB, not GRB... but that might explain why the flicker is always green that I see on the first pixel!  I thought it was picking up the second byte, but it is the first..

So i would need to change the order of bytes... the plot thickens... sorry to be asking you so much!  

I don't really understand what this bit of your example does..
```
    if (pixels[I].R == 255) {
        pixels[I].R = 0;
    }
    else {
        pixels[I].R++;
    }
```
actually i don't get any of it!  as much as i've stared at it!  
pixels is the GRB struct.  so this holds the 3 bytes of colour but in the order required to go out to the pixels.  

Then you cast this to you pixel buffer using ```GrbPixel* pixels = (GrbPixels*)NeoPixelBus.Pixels();```  Could you explain how the pointer and cast works here? 

then the ```pixels[I].R == 255``` says if the red component is maximum, set it to 0, otherwise add one...  I think i just got it...  this walks each red pixel making it brighter... you've just given me an example effect!  (correct me if I'm wrong)  I might have just got it! 

However, can you do this without a struct?  

````
uint8_t* pixels = (uint8_t*)NeoPixelBus.Pixels();
for (int I = 0; I < NeoPixelBus.PixelCount; i =+ 3) {
buffer[I] = pixels[I+1]       // Red
buffer[I+1] = pixels [I]       // Green
buffer[I+2] = pixels [I+2 ]      // Blue
}

Makuna, 2015-05-27:
Its just a demonstration of how to reference the raw bytes buffer as pixels; it really is just showing how to know which bytes are which pixels and which color component.

Sent from Windows Mail

From: sticilface
Sent: ‎Wednesday‎, ‎May‎ ‎27‎, ‎2015 ‎7‎:‎45‎ ‎AM
To: Makuna/NeoPixelBus
Cc: Makuna@hotmail.com

Actually... forget that. My bytes are RGB, not GRB... but that might explain why the flicker is always green that I see on the first pixel! I thought it was picking up the second byte, but it is the first..

So i would need to change the order of bytes... the plot thickens... sorry to be asking you so much! 

I don't really understand what this bit of your example does..
    if (pixels[I].R == 255) {
        pixels[I].R = 0;
    }
    else {
        pixels[I].R++;
    }

actually i don't get any of it! as much as i've stared at it!

pixels is the GRB struct. so this holds the 3 bytes of colour but in the order required to go out to the pixels. 

Then you cast this to you pixel buffer using GrbPixel* pixels = (GrbPixels*)NeoPixelBus.Pixels(); Could you explain how the pointer and cast works here? 

then the pixels[I].R == 255 says if the red component is maximum, set it to 0, otherwise add one... I think i just got it... this walks each red pixel making it brighter... you've just given me an example effect! (correct me if I'm wrong) I might have just got it! 

However, can you do this without a struct? 
uint8_t* pixels = (uint8_t*)NeoPixelBus.Pixels();
for (int I = 0; I < NeoPixelBus.PixelCount; i =+ 3) {
buffer[I] = pixels[I+1]       // Red
buffer[I+1] = pixels [I]       // Green
buffer[I+2] = pixels [I+2 ]      // Blue
}

Makuna, 2015-05-27:
These three concepts you should gain some familiarity with if you plan to access lower level memory buffers.  They are crucial to build easier to read and fast routines.

Sent from Windows Mail

From: sticilface
Sent: ‎Wednesday‎, ‎May‎ ‎27‎, ‎2015 ‎6‎:‎54‎ ‎AM
To: Makuna/NeoPixelBus
Cc: Makuna@hotmail.com

Thank you! However, in that one example you've introduced two concepts that I'm not familiar with at all really. well maybe even 3. 

1) struct
2) casting
3) (pointers.. although i know a bit about them)

Your code takes an RGB object and casts it to the buffer directly using a pointer? (is that right)? 

Now.. if i have a buffer already... that is filled with G1, R1, B1, G2, R2, B2, G3, R3, B3, ...
say for example a UDP packet... or a buffer filled from serial... can I just do something like
 for (int I = 0; I < NeoPixelBus.PixelCount; I++) {
pixelbuf[I] = [insert something here];
}

or do you have to define a struct, and cast to it... ?

closed 2015-06-09 by sticilface