Makuna/NeoPixelBus #10: Examples only work using GPIO13

einsteinx2, 2015-05-26:
For some reason, I can only get the examples to control the WS2812 LEDs when connected to GPIO13 on my ESP12 (4 Mbyte model).

Any other pin and while it programs the ESP module successfully, the LEDs do nothing.

IIRC GPIO13 is the SPI pin on the ESP. Does the NeoPixelBus library handle GPIO13 differently than other pins?

Makuna, 2015-05-26:
I just picked up a ESp-12 this last week-end.  I just gave it a try, and while it worked on D7 (Gpio13), D6 (Gpio14), and D10 (Gpio01), the colors where not correct.  Further, it REQUIRED the resister inline to even work. 

Did you try it with the resistor as defined by Adafruits pixel ring requirements?

I am investigating deeper as I see incorrect colors.

einsteinx2, 2015-05-27:
I have no resistor on the data pin, didn't realize that was a requirement. Seems like it was correct colors.

Makuna, 2015-05-27:
Ok, I had a bad connector in my breadboard, so ground between esp8266 and the NeoPixel was floating, SO this caused the incorrect colors.  Once I fixed this, I was able to work on a sample of the pins which included the D7, D6, and D10.

The resistor is only needed if you your NeoPixel "collection" doesn't already have one.  AdaFruit rings are an example.  RAW WS2812 are another.  Many of the little boards that have a WS2812 on them already have the resistor. Further, its not required to work, it just helps "save" the first pixel from being burned out.

How are you connecting the IO pin between the NeoPixel and the Esp8266?  Level change chip?  Directly?  What voltage are you running the NeoPixels at?

The esp8266 is 3.3v, WS2812 require the IO to be within +-0.5v of VCC and they have a minimum of VCC of 3.5v.  If you don't drop the VCC to below 3.8v or have a level change chip, then you will get erratic or no changes.

einsteinx2, 2015-05-31:
I'm using WS8212 strips purchased from eBay (not AdaFruit). I'm running them off of a 3.3v power supply (which is also powering the ESP-12 module). I'm successfully using 31 LEDs without issue off of GPIO13 using this setup. 

No resistor on the data line, though I'll add one based on the NeoPixel guide. I'll try other pins again after adding the resistor.

einsteinx2, 2015-05-31:
Ah looks like GPIO5 and 4 are mislabeled, turns out I'm able to use them both! Wasn't able to use GPIO16 though (perhaps that's also mislabeled?). Anyway, great stuff!

Not sure if the resistor made any difference, I think it was the mislabeling that threw me off. Looks like this is resolved then!

closed 2015-05-31 by einsteinx2