Makuna/NeoPixelBus #14: WDT reset when using neopixel examples

CrazyHarold, 2015-06-04:
I have installed the neopixelbus lib, but when i try the neopixelbus examples, my ESP8266 issues a WDT reset every second or so.

Any ideas ?
I am using the latest arduino 1.6.4 ide and esp toolchain from https://github.com/esp8266/Arduino

Regards, Harold.

Makuna, 2015-06-04:
Help me out with more details.  

Which module?  
What pin did you connect the neopixels to?  What pin does the example use (include the line of code)?
What voltage are you running the NeoPixels at? 
Are you using a level converter on the io pin between the esp8266 and the NeoPixels?
By "esp toolchain" you mean you installed the esp8266 support for Arduino using the BoardManager and not the older install or custom build?  What version is it showing in the BoardManager?

CrazyHarold, 2015-06-04:
Hi Makuna, 

Module ESP12 (NodeMCU dev board)
I did not yet connect any neopixel strips. Just compiled the neopixelfun / test examples and uploaded them to the board. After that, the board reset's every few seconds and spits the following text on my serial port.

 ets Jan  8 2013,rst cause:4, boot mode:(3,6)

wdt reset
load 0x40100000, len 28688, room 16 
tail 0
chksum 0xbc
load 0x3ffe8000, len 1332, room 8 
tail 12
chksum 0xd6
ho 0 tail 12 room 4
load 0x3ffe8540, len 1508, room 12 
tail 8
chksum 0x72
csum 0x72
rd

Board manager shows
esp8266 by ESP8266 Community version 1.6.4-673-g8cd3697 INSTALLED

Makuna, 2015-06-04:
Do you have NodeMCU v1 or v2?

If you have V2, did you manually apply https://github.com/esp8266/Arduino/issues/292 to your install?

CrazyHarold, 2015-06-04:
Hi Makuna,

Thanks for your help !
I am running devkit v0.9.

My "normal" example programs run fine. 
Eg. the WiFiCLient example is working fine.

However : I just changed the qio to dio, but it did not change anything.

Regards Harold.

Makuna, 2015-06-04:
The version number is inconsistent, mine states the same but it's a V2 board.  

Make sure you still pick a  valid pin.  Which pin are you setting? Both samples?  

I will try to repo.

CrazyHarold, 2015-06-04:
Hi Makuna,

Default in the example the pin was set to "8".
I have switched to "2" now (gpio2 ?) and it appears to be not resetting anymore (so it looks like it works now).
I will try the gpio2 pin with my neopixel strip next week.

Thank you VERY much for your support !

Regards, Harold.

closed 2015-06-04 by CrazyHarold

wishwamithra, 2018-04-05:
Hello, were you able to drive Neopixel strip? I added some debug lines (Serial.println(...)) and found out that it crashes right when you call begin() function of the Adafruit_NeoPixel object. I crashed on me when I set up Pin 6. But as you said it doesn't crash when I switch to Pin 2. But it doesn't light up the LED strip. Probably because LED strip's data line needs 5V, where as NodeMCU's output is 3.3V. But doesn't LED strip's data line consider 3.3V as a high? Appreciate if someone could let me know.

Makuna, 2018-04-06:
@wishwamithra This is about the NeoPixelBus not Adafruit_NeoPixel library.  And the NeoPixels (actually LEDs) vary with support for 3.3v data line.  The specs state 3.8v minimum; but some work fine.  See the Wiki for my library for the FAQ that covers this topic.

wishwamithra, 2018-04-07:
Hi Makuna,

Thank you very much for the update. I know the following question is not relevant to this thread, but do you have a working ESP8266 code for me to upload new sketches OTA?

Thanks,
Wishwa.

Makuna, 2018-04-07:
It has been working very well for two years; so i am confused by your question.  Are you running into a problem?  if so, jump on the gitter channel (see Wiki) and ask away.

But please do not continue to use this issue for more discussion.

debadalona, 2022-02-18:
Hello,
The same has happened to me. I have taken the example of neopixel and it is reset by watchdog timer.
If I change the port from 6 to 2, it does not reset but it does not work, it does not turn on the leds.

The only thing that happens is that in the sketch you must put:
#define PIN D6 and not #define PIN 6

It's just that....