Makuna/NeoPixelBus #16: Error compiling

andySigler, 2015-06-08:
Below is a printout from the IDE. I would try fixing, but I'm lost by your syntax :)

```
/Users/andy/Documents/Arduino/libraries/NeoPixelBus/NeoPixelBus.cpp: In lambda function:
/Users/andy/Documents/Arduino/libraries/NeoPixelBus/NeoPixelBus.cpp:723:18: error: expected '{' before string constant
         [lo]     "r" (lo));
                  ^
/Users/andy/Documents/Arduino/libraries/NeoPixelBus/NeoPixelBus.cpp: In member function 'void NeoPixelBus::Show()':
/Users/andy/Documents/Arduino/libraries/NeoPixelBus/NeoPixelBus.cpp:723:18: error: expected ';' before string constant
Error compiling.
```

I was originally going to try this on the esp8266, but I get an even longer list of compile errors when selecting that board...

```
/Users/andy/Documents/Arduino/libraries/NeoPixelBus/NeoPixelBus.cpp: In function 'void send_pixels_800(uint8_t*, uint8_t*, uint8_t)':
/Users/andy/Documents/Arduino/libraries/NeoPixelBus/NeoPixelBus.cpp:88:41: error: '_BV' was not declared in this scope
     const uint32_t pinRegister = _BV(pin);
                                         ^
/Users/andy/Documents/Arduino/libraries/NeoPixelBus/NeoPixelBus.cpp:94:20: error: 'GPIO_OUT_W1TC_ADDRESS' was not declared in this scope
     GPIO_REG_WRITE(GPIO_OUT_W1TC_ADDRESS, pinRegister);
                    ^
/Users/andy/Documents/Arduino/libraries/NeoPixelBus/NeoPixelBus.cpp:94:54: error: 'GPIO_REG_WRITE' was not declared in this scope
     GPIO_REG_WRITE(GPIO_OUT_W1TC_ADDRESS, pinRegister);
                                                      ^
/Users/andy/Documents/Arduino/libraries/NeoPixelBus/NeoPixelBus.cpp:96:23: error: 'class EspClass' has no member named 'getCycleCount'
     cyclesStart = ESP.getCycleCount() + CYCLES_800;
                       ^
/Users/andy/Documents/Arduino/libraries/NeoPixelBus/NeoPixelBus.cpp:112:35: error: 'class EspClass' has no member named 'getCycleCount'
                 cyclesStart = ESP.getCycleCount();
                                   ^
/Users/andy/Documents/Arduino/libraries/NeoPixelBus/NeoPixelBus.cpp:117:28: error: 'GPIO_OUT_W1TS_ADDRESS' was not declared in this scope
             GPIO_REG_WRITE(GPIO_OUT_W1TS_ADDRESS, pinRegister);
                            ^
/Users/andy/Documents/Arduino/libraries/NeoPixelBus/NeoPixelBus.cpp:122:29: error: 'class EspClass' has no member named 'getCycleCount'
                 while ((ESP.getCycleCount() - cyclesStart) < CYCLES_800_T1H);
                             ^
/Users/andy/Documents/Arduino/libraries/NeoPixelBus/NeoPixelBus.cpp:126:29: error: 'class EspClass' has no member named 'getCycleCount'
                 while ((ESP.getCycleCount() - cyclesStart) < CYCLES_800_T0H);
                             ^
/Users/andy/Documents/Arduino/libraries/NeoPixelBus/NeoPixelBus.cpp: In function 'void send_pixels_400(uint8_t*, uint8_t*, uint8_t)':
/Users/andy/Documents/Arduino/libraries/NeoPixelBus/NeoPixelBus.cpp:141:41: error: '_BV' was not declared in this scope
     const uint32_t pinRegister = _BV(pin);
                                         ^
/Users/andy/Documents/Arduino/libraries/NeoPixelBus/NeoPixelBus.cpp:147:20: error: 'GPIO_OUT_W1TC_ADDRESS' was not declared in this scope
     GPIO_REG_WRITE(GPIO_OUT_W1TC_ADDRESS, pinRegister);
                    ^
/Users/andy/Documents/Arduino/libraries/NeoPixelBus/NeoPixelBus.cpp:147:54: error: 'GPIO_REG_WRITE' was not declared in this scope
     GPIO_REG_WRITE(GPIO_OUT_W1TC_ADDRESS, pinRegister);
                                                      ^
/Users/andy/Documents/Arduino/libraries/NeoPixelBus/NeoPixelBus.cpp:149:23: error: 'class EspClass' has no member named 'getCycleCount'
     cyclesStart = ESP.getCycleCount() + CYCLES_400;
                       ^
/Users/andy/Documents/Arduino/libraries/NeoPixelBus/NeoPixelBus.cpp:162:35: error: 'class EspClass' has no member named 'getCycleCount'
                 cyclesStart = ESP.getCycleCount();
                                   ^
/Users/andy/Documents/Arduino/libraries/NeoPixelBus/NeoPixelBus.cpp:166:28: error: 'GPIO_OUT_W1TS_ADDRESS' was not declared in this scope
             GPIO_REG_WRITE(GPIO_OUT_W1TS_ADDRESS, pinRegister);
                            ^
/Users/andy/Documents/Arduino/libraries/NeoPixelBus/NeoPixelBus.cpp:171:29: error: 'class EspClass' has no member named 'getCycleCount'
                 while ((ESP.getCycleCount() - cyclesStart) < CYCLES_400_T1H);
                             ^
/Users/andy/Documents/Arduino/libraries/NeoPixelBus/NeoPixelBus.cpp:175:29: error: 'class EspClass' has no member named 'getCycleCount'
                 while ((ESP.getCycleCount() - cyclesStart) < CYCLES_400_T0H);
                             ^
Error compiling.
```

Makuna, 2015-06-09:
Looks like I had a merge problem, investigating.

Makuna, 2015-06-09:
Hey Andy, I fixed the merge issue with AVR builds.  These should be working now.

For esp8266, are you using the official release using the Board Manager install?  The missing _BV macro was fixed in the esp8266/arduino for the board manager release.

andySigler, 2015-06-09:
Thanks!!

Yes I'm using the Board Manager, using [this json file](http://arduino.esp8266.com/package_esp8266com_index.json)

If it's a problem with the esp8266 support, I'll move my issues elsewhere. This is all diving below my understanding, so thank you for the help!

Makuna, 2015-06-19:
Andy, did you get this working?  Please join the chat and we can quickly go back and forth to get this straightened out.

https://gitter.im/Makuna/NeoPixelBus

Makuna, 2015-07-06:
Closing this as I suspect it was issue that got resolved.

closed 2015-07-06 by Makuna