Makuna/NeoPixelBus #17: ESP8266 drives LEDs slow

drom, 2015-07-06:
Great library! Thank you!
Running your examples on ESP-01 board with Arduino 1.6.5 on the Linux. Timing on the oscilloscope looks way slower (10-20 times) then NeoPixels requires. Not sign of Light. Any suggestions?

Makuna, 2015-07-06:
What times are you measuring on the scope?  
A single bit should be 12.5us, each pixel takes 24 bits,  then times the number of pixels on the bus.  Then between refresh of the bus requires a 50 us seperation.

drom, 2015-07-06:
AH! I was trying to drive my WS2812 with it!
https://learn.sparkfun.com/tutorials/ws2812-breakout-hookup-guide/ws2812-overview
Sorry for the stupid question, but aren't NeoPixels and WS2812 the same?

Makuna, 2015-07-06:
Yes they are, and the guide seems correct.

drom, 2015-07-06:
Then I expect to see (1.15 ~ 1.3)[us] per pixel. Should I?
Bit0 = (0.35 + 0.8 = 1.15) [us]
Bit1 = (0.7 + 06 = 1.3)[us]

Makuna, 2015-07-06:
Not 1.25us per pixel, it's per bit.  It takes 24 bits to represent a pixel.  It states this in the link you gave.

drom, 2015-07-06:
yes, 1.25us per bit. And on oscilloscope I see 10 times of this.

Makuna, 2015-07-06:
Which branch are you using?  Master or NeoPixelAnimator?  For esp8266, I recommend NeoPixelAnimator branch.
How many pixels are you running?  
Just one instance of NeoPixelBus?

I have tested this on a ESP-01 and ESP-12e and both give the same timing.  If you are correctly seeing pixels change to the colors you expect, then the timing can't be off as they only accept a small amount of timing error before they start doing strange things.  An error of 0.150us per bit is documented as acceptable, but I have found that if its kept below 0.3us per bit it still works.  They truly will not function much  beyond this.

Can you capture your oscilloscope image?  If its not obvious on the image, please also include scale for time and voltage.

drom, 2015-07-07:
Thank you for the help. After going back and forth It works now.
Strange slow signal disappeared. I suspect it was UART signal or something other interference.

closed 2015-07-07 by drom