Makuna/NeoPixelBus #18: esp12 crashes in accespoint if strip.show() called.

Duality4Y, 2015-08-01:
Hi there!
I am creating code for my esp12 using arduino,
and your library to address ws2812 ledstrips.
it is connected to pin 14.

when i comment strip.show() my html shows and the esp12 doesn't crash.
but if i uncomment strip.show() my esp12 crashes in accespoint mode when i try and reach my html pages.

code:
https://github.com/Duality4Y/Sketchbook/tree/master/sketchbook/webserver-esp

in ws2812.cpp you can see that the update function calles the strip.show();
and in stripcontrol.cpp in hanldeStrips(),
you'll see i Uncommented the update function.

so any ideas? 

by crashing I meen it triggers the WDT.

Makuna, 2015-08-02:
Are you using the default branch of my library?  For esp8266, there is another branch with better memory handling with the animation supported separated out.

First, try a different pin, like  4 or 5 that doesn't share functionality.  Just to make sure there isn't something else being effected by the pin change.  14 is the clock for one of the SPI buses.

150 pixels is 4ms to just send out the pixel data.  For testing, what happens if you just make it 8 pixels?  Does it still repro?  This shouldn't be a problem but just something to check.

The yield in your loop() isn't needed,

Duality4Y, 2015-08-02:
I can't change pins i am stuck with the hardware i got :)
branch i am using:
```
robert@laptop:~/Sketchbook/sketchbook/libraries/NeoPixelBus$ git branch -a
* NeoPixelAnimator
  master
  remotes/origin/HEAD -> origin/master
  remotes/origin/MultiBus
  remotes/origin/NeoPixelAnimator
  remotes/origin/master
```

I know the yield in the loop isn't needed :) but i thought why not.
I'll try and make it 8 pixels and see what happens :)

Duality4Y, 2015-08-02:
tried 8 pixels still crashes,
tried pin 4 and crashes even faster.

Duality4Y, 2015-08-13:
Ok I have to strip connected top pin 14 and that works 150pixels works.
does still crash in ap mode, did:
if apmode then dont run.

closed 2015-08-13 by Duality4Y