Makuna/NeoPixelBus #19: Wrong colors with ESP-12-E module

apullin, 2015-08-15:
I am using this library with an ESP-12-E module, as part of a NodeMCU-1.0 board.
Running the stock example that is meant to illuminate 4 pixels, I am getting wrong color output on the pixels, where I get: blue, red, black, white, green on the first 5 pixels.

The setup is driving two chained strips of 144 pixel/m strips, brought from Adafruit.
The strip is declared to operate with pin D2 (where 'D2' is used in the code). Same result when driving with D1.

Makuna, 2015-08-15:
Depending on the actual pixels, you may need to include the flag to change to rgb.  There is a comment on this.  Please check if this fixes the issue for you.

I suggest those who use use the esp to use the animator branch, it has a more flexible animation support that can't be supported on the avr chips.

Note, I also use the 12e as my priMary device for testing.

apullin, 2015-08-15:
The flag does not fix the issue, and only switched the color of the first two pixels, giving R,B,black,W,G.

Makuna, 2015-08-15:
How are you powering the NeoPixel and how are you connecting them?  
Which model of pix3ls?
What version of the EspArduino?

apullin, 2015-08-17:
OK, this was entirely a mistake on my part. I was not using a 3.3-to-5 level shifter. Code works with circuit corrected.

closed 2015-08-17 by apullin