Makuna/NeoPixelBus #21: Setting colors during setup causes white flashes

Makuna, 2015-08-24:
from @sticilface 

example app
```
#include <NeoPixelBus.h> // NeoPixelAnimator branch

#define pixelCount 7 // Strip has 30 NeoPixels
#define pixelPin 2 // Strip is attached to GPIO2 on ESP-01

NeoPixelBus strip = NeoPixelBus(pixelCount, pixelPin);

void setup() {

    strip.Begin();
    strip.Show();

   // delay(100);

for (int i = 0; i < 5; i++) {
  strip.ClearTo(RgbColor(0,0,0));
  strip.Show();
  delay(100);
  strip.ClearTo(RgbColor(0,0,255));
  strip.Show();
  delay(100); 
}

}

void loop() {

  delay(10);
  strip.ClearTo(RgbColor(0,0,0));
  strip.Show();
  delay(500);
  strip.ClearTo(RgbColor(0,255,0));
  strip.Show();
  delay(500); 
}
```

sticilface, 2015-08-24:
This sketch works fine btw using the UART Serial method :) https://github.com/Makuna/NeoPixelBus/pull/22

but does not work with big banging.

Makuna, 2015-12-29:
I believe this was due to the issue of the actual chip was running at 80mhz during setup even when you select 160mhz.  With the latest versions of espArduino this was fixed so that setup runs at the speed you select.

sticilface, 2015-12-29:
yes correct!

closed 2015-12-29 by Makuna