Makuna/NeoPixelBus #26: Strange behaviour with Animation branch on NodeMCU arduino ide

madshobye, 2015-09-08:
When using the animation branch it seems like the timing is off - the lights are glitchy and mustly white. The other branch looks more tight. I have used it where it was ok. But after switching to the staging version of arduino/esp ide library it became glitchy.

http://arduino.esp8266.com/staging/package_esp8266com_index.json

I am not sure what det difference is - just giving you a heads up. FYI: I had to chose nodemcu 0.9 board to be able to upload. Before i just used the generic esp board from arduino.

This is the branch i used to use without your library glitching.

http://arduino.esp8266.com/package_esp8266com_index.json

Makuna, 2015-09-09:
Does your project have active wifi?

madshobye, 2015-09-10:
Nope. Just basic testing using your example. A detail i renemeber now. I
only tested the non animation version on the first branch. e.g. All I know
is that the animation version does not work well with nodemcu.

Makuna, 2015-09-10:
Its not the NodeMCU, this is what I use, I am investigating, but there are several issues it could be.
1) have you ever had it working well?  If not, then make sure you either use a level adjuster for the IO pin if you power the pixels with 5v (neopixels require IO to be within +-0.5v to vcc), or you must supply power that that both can use, 3.8v (esp is 3.3v, NeoPixels min is 3.8v)
2) There is a change happening in how the data is written out due to esp SDK change that makes turning off interrupts not possible anymore.  The esp/Arduino code make have changed noInterrupts to support the new SDK which would have broken my code.  I am addressing this soon in the animator branch, and there will be a new branch that will be required if you plan to have WiFi.

madshobye, 2015-09-15:
I use vin 5v from usb. The data is 3.3v directly from the pin. I am not
sure this answers your question. I might have to go back to the old branch
to verify.

Makuna, 2015-09-15:
Ok, your problem is how you have connected the hardware.  This is a common issue and several of the closed issues are the same problem.  

You can't drive the NeoPixel at 5v and also use 3.3v data line.  The data line MUST be within 0.5v of the vcc.  

Thus, either you need to either run both the esp and neopixels at a common usable voltage level (3.8v) or you will need level shifters to convert the 3.3v IO from the esp to 5v for the esp (these can be hade for a few dollars from most places that also sell the esp)

madshobye, 2015-09-18:
Okay I have made a level shifter with two not gates in series.

It is strange - because my own code works (a modified version of your code
- se below) but the examples provided with the animation library still yels
strange results.

#include <NeoPixelBus.h>

#define pixelCount 200

NeoPixelBus strip = NeoPixelBus(pixelCount, 2);
uint16_t effectState = 0;

float h = 0;
float offset = 0;

void setup()
{
  strip.Begin();
  strip.Show();

}

void loop()
{ h = 0;
  for(int i =0; i < pixelCount; i ++)
  {
    h = h + 1.0f;

    strip.SetPixelColor(i, HSLtoRGB( round(h + offset)% 360,100,30));
  }
   offset = offset + 10.0f;
  strip.Show();

}

// This function converts the "color" object to the equivalent RGB values of
// the hue, saturation, and luminance passed as h, s, and l respectively

RgbColor HSLtoRGB(float h, const float s, float l)
{
  int tmpR,tmpG,tmpB;
  unsigned int r = 0;
  unsigned int g = 0;
  unsigned int b = 0;

  float L = ((float)l)/100;
  float S = ((float)s)/100;
  float H = ((float)h)/360;

  if(s == 0)
  {
    r = l;
    g = l;
    b = l;
  }
  else
  {
    float temp1 = 0;
    if(L < .50)
    {
      temp1 = L*(1 + S);
    }
    else
    {
      temp1 = L + S - (L*S);
    }

    float temp2 = 2*L - temp1;

    float temp3 = 0;
    for(int i = 0 ; i < 3 ; i++)
    {
      switch(i)
      {
      case 0: // red
        {
          temp3 = H + .33333f;
          if(temp3 > 1)
            temp3 -= 1;
          HSLtoRGB_Subfunction(r,temp1,temp2,temp3);
          break;
        }
      case 1: // green
        {
          temp3 = H;
          HSLtoRGB_Subfunction(g,temp1,temp2,temp3);
          break;
        }
      case 2: // blue
        {
          temp3 = H - .33333f;
          if(temp3 < 0)
            temp3 += 1;
          HSLtoRGB_Subfunction(b,temp1,temp2,temp3);
          break;
        }
      default:
        {

        }
      }
    }
  }
  tmpR = round((((float)r)/100.0)*255.0);
  tmpG= round((((float)g)/100.0)*255.0);
  tmpB = round((((float)b)/100.0)*255.0);
  return RgbColor(tmpR,tmpG,tmpB);

}
static void HSLtoRGB_Subfunction(unsigned int& c, const float& temp1, const
float& temp2, const float& temp3)
{
  if((temp3 * 6) < 1)
    c = (unsigned int)((temp2 + (temp1 - temp2)*6*temp3)*100);
  else
    if((temp3 * 2) < 1)
      c = (unsigned int)(temp1*100);
    else
      if((temp3 * 3) < 2)
        c = (unsigned int)((temp2 + (temp1 - temp2)*(.66666 -
temp3)*6)*100);
      else
        c = (unsigned int)(temp2*100);
  return;
}

Makuna, 2015-09-18:
Please be specific.  Which sample?  What isn't working?  What code did you replace?

The HSLColor object either takes uint8_t values between 0 and 255 or if you change to use floats, it will require values between 0.0 and 1.0.  Your code above is assuming 0-360 for H, and not sure about the others.

Makuna, 2015-10-03:
If you are using the UART branch, then you will to adjust the running speed of the esp to 160mhz.  There is timing issue with the hardware uart that shows up at 80mhz due to the way the code is using it.

Makuna, 2016-01-23:
Note, there are now two WiFi compliant branches, UartDriven and DmaDriven, with the latest IDE and EspArduino they no longer are sensitive to the esp speed (the issue got fixed).  Please let me know if you are still seeing problems.

madshobye, 2016-01-23:
I have used the Uardriven and i works like a charm even over wifi.

http://fablab.ruc.dk/controlling-neopixels-with-processing-via-wifi-to-nodemcu/

closed 2016-01-24 by Makuna

vlast3k, 2016-01-27:
Sorry for commenting on the closed issue, but could you briefly explain in which scenarios the master branch does not work? With any wifi communication, or just when something specific is done?
Somehow with the UART-driven branch - the blue led on the esp is always on, and with the DMA driven, i cannot use the Serial receve.

Makuna, 2016-01-27:
feel free to create a new issue.