Makuna/NeoPixelBus #28: Unstable behavior with 50+ pixels

leonyuhanov, 2015-09-16:
I have a NodeMCU V1.0 module
I can get the UART based library to work using pin 4(the only pin that works)
However it only drives about 40-50 pixels. anything greater and they start flickering.
Things i have tested for:
-I ran the strip using both 5v and 3.3v but the same issue occurs on both voltages
-I'm getting a nice steady 3.3v signal from Pin 4, so when i have less than 50 pixels its all good
-im running the strip on very low max values, so that the saturation level is as an example (10,10,10)

leonyuhanov, 2015-09-16:
I have just changed my IDE to compile to 160Mhz instead of 80 and its working fine now...

closed 2015-09-16 by leonyuhanov

Makuna, 2015-09-16:
You can't run neopixels at 3.3v reliably.  Their min voltage is 3.8v. The effects you see is what happens.  They also require the IO pin to be within +-0.5v of the vcc.
I suggest getting a level converter to adjust the IO from 3.3v to 5.0v.

leonyuhanov, 2015-09-16:
Im curretly running my strip at 5v with no problems
Leon