Makuna/NeoPixelBus #29: Uart example code does not work with ESP8266 v12

Richard210363, 2015-10-03:
The example code, NeoPixelTest from the Uart branch, does not work with an ESP8266 v12 coupled with an AdaFruit NeoPixel ring.

I get no output from the LEDs at all.
Using the Master branch works fine with NeoPixelTest 
I removed the HslColor code when doing both tests.

I can't get the UART branch to light an LED at all.

Makuna, 2015-10-03:
Are you running your esp at 80mhz? There is an issue where the UART version must run at 160mhz. This is noted in the Readme.

You can set this in the boards menu. There are been no effects noted when running at 160mhz rather than 80mhz.

Let me know if that fixes your issue.

Since you have them working with the other branch, then you are either using level shifters for the IO or you are powering your pixels with near 3.8v. If you are not, note that the pixels require the data line to be within +-0.5v of their Vcc to run reliably.

Richard210363, 2015-10-05:
I've tried the 160MHz option and my NeoPixels lit!
This is the with a 12 LED ring. 

Unfortunately, when I added more rings ( I need 5) only 1 ring lit up 

The rings are not cascading to each other.  
Each ring is on a separate pin.

It looks like when I do this:

NeoPixelBus strip_04 = NeoPixelBus(12, 04);
NeoPixelBus strip_05 = NeoPixelBus(12, 05); 
NeoPixelBus strip_15 = NeoPixelBus(12, 15); 
NeoPixelBus strip_02 = NeoPixelBus(12, 02); 
NeoPixelBus strip_14 = NeoPixelBus(12, 14);

strip_04.Begin();
strip_05.Begin();
strip_15.Begin();
strip_02.Begin();
strip_14.Begin();  

strip_04.Show();
strip_05.Show(); 
strip_15.Show();
strip_02.Show();
strip_14.Show();

that only pin 014 is sending data in a way the NeoPixel can read it.

I'm running the NeoPixels and the ESP8266 at 3.7V.

All the above works fine when I use the non UART library :-)

I'm going to stick with the working library for now as I have a deadline but if you want me to test anything, let me know.

Makuna, 2015-10-05:
I do believe this is mentioned in the readme but it may not be clear.  The uart version is limited to to only one pin.  There is no solution to this as its a hardware feature that supports only one pin.

There maybe a way to jumper the out of the last pixel on the ring to the in on the next and thus chain the rings together.

Richard210363, 2015-10-06:
Ah, this is the phrase "For the exp8266, only the TXD1 pin is supported due to the hardware uart restriction and this argument is ignored."

I'll look at chaining but for now I'l stick with the non UART library as it is doing what I require.

Thanks for answering my queries so fast.

emcniece, 2015-10-25:
I can't get any action to show up either. The object definition in the uart branch sets up like this:

```
NeoPixelBus strip = NeoPixelBus(4, 8, NEO_RGB | NEO_KHZ800);
```

... which sets the output pin to 8. What pin number (or more importantly, digital output number) does TXD1 sit on?

Makuna, 2015-10-25:
@emcniece the pin you pass is ignored for uart branch.

The actual pin depends on which esp you have.  The esp 12 e this is the gpio2 pin, labeled as d4 on the dev board.

emcniece, 2015-10-25:
Thanks for the clarification @Makuna! [This](https://encrypted-tbn2.gstatic.com/images?q=tbn:ANd9GcQJ-5QX3jWKFPoTkJ7ATM7noQIM0gy8G7RqMNkIZEWyRA_8QAl6Jw) is the board I'm working with - I'll give the other pins a whirl...

NickWaterton, 2015-12-09:
I have the esp V12 (no e that I can see). Using the uart branch I get either a lot of randomly flashing pixels, or the chip constantly resets, depending on which example test I run.

I am testing at 3.7V, on gpio2 at 160 MHz.

The master branch works fine, but I need wifi, and the chip randomly resets after a few minutes using 19 pixels -don't know if this is due to wifi or not.

Any ideas as to why the uart branch gives such weird results? Is it the V12? I'm assuming gpio2 is correct for this chip, as I do get randomly flashing pixels.

thanks.

Richard210363, 2015-12-09:
I noticed that above 15 pixels I had resets with the Master branch.
I got round it by using more GPIOs and limiting each one to 12 pixels.  Not had a problem since.

NickWaterton, 2015-12-10:
Tried it with 12 pixels, same thing.

Any ideas about the uart branch? Can't get anything sensible out of it, but I'm not totally sure I'm using the right pin.

I've tried other pins, but gpio2 is the only one that produces anything, scrambled though it is.

NickWaterton, 2015-12-10:
Update,

I have it working! With 19 pixels (at 160MHz, but it did work at 80).

Could not get the examples to work at all, which was causing me to think that something fundamental was wrong.

I think in fact the examples do not actually work (at least on my hardware). 

I have the right pin (gpio2), and finally extracted the UART code and inserted it in a modified neopixel library.

I have now had it running with wifi, mqtt, OTA updates and 19 neopixels changing state every 5 seconds (via mqtt) for over 1hour with no resets or other issues.

OTA updates remains untested, as the Arduino IDE seems to be broken in this respect (I'll try Python updates later), but the hooks are in there.

Thank you for the UART code!

NickWaterton, 2015-12-11:
Update. Still running 5 hours later. Will run overnight just to satisfy myself this is a complete solution.

NickWaterton, 2015-12-12:
Ok all is working after 24 hours. OTA updates via Python and http work (http is a bit hit and miss).

The UART code is the solution for wifi enabled esp8266 chips.

NickWaterton, 2015-12-12:
I decided to dig into the example code for the UART branch, to try to figure it out. There seem to be some simple coding errors which prevent them working.

For example in "loop around" the main loop is:
 for (uint16_t offset = 5; offset >= 0; offset--)

This cannot work as the variable is defined as a uint so the test will always be true, the loop will never exit and offset wraps round to 64k.

Please beware of these traps!

vindolin, 2015-12-20:
@NickWaterton so what exactly did you change to get it to work? All my pixels light up in the NeoPixelFun example with some of em flickering from time to time. (NodeMCU, GPIO2, 160 MHz).
The NeoPixelTest example works as expected.

NickWaterton, 2015-12-21:
Once I figured out that they did not (and could not) work, I didn't pursue them any further, but substituted my existing working neopixel routines.

One of the problems was that the library not only implemented the uart code, but also various animations, which were hard to figure out with no documentation (or working examples).

I extracted the uart code, inserted it into Adafruit's standard neopixel library (along with some useful routines for fading, RgbColor type and so on), and used the regular neopixel routines I already had to test that it worked.

The real trick after that to make it work with an esp8266 was to avoid for loops and use of delay (where I could). Using millis (or micros) timing to control loops works much better, but is much harder to code.

The animations built in to Makunas library are probably intended to make this easier, but I was on a steep learning curve, so one thing at a time!

Nick Waterton P.Eng

Unsubscribe from our commercial electronic messages.
http://supportcentral.ge.com/esurvey/takesurvey.asp?p=17778&d=3834102
D�sabonner de nos messages �lectroniques commerciaux.
http://supportcentral.ge.com/esurvey/takesurvey.asp?p=17778&d=3839563

Makuna, 2015-12-29:
@NickWaterton Yes, my animation library was meant to help users in doing animations without the need for delay().  Basically it makes a callback on an arbitrary time to have you do your animation.  The timing value is a an abstract 0.0 to 1.0 progress from start to end of the animation.  You use this to calculate the color you want at that time.  The included blend functions can take this value directly.  In one of the samples it uses a lambda to define the callback inline, 

```
        // define the effect to apply, in this case linear blend
        AnimUpdateCallback animUpdate = [=](float progress)
        {
            // progress will start at 0.0 and end at 1.0
            RgbColor updatedColor = RgbColor::LinearBlend(originalColor, color, (uint8_t)(255 * progress));
            strip.SetPixelColor(pixel, updatedColor);
        };
```

Makuna, 2015-12-29:
I have updated the uartdriven branch to some changes that should improve the accuracy.  I have examined a capture of the data for 300 leds (I don't actually have that many to test) and I am not seeing any problems in the capture.  
If you see issues, please disable wifi features and see if that changes anything.

probonopd, 2016-01-25:
@NickWaterton 

> I have now had it running with wifi, mqtt, OTA updates and 19 neopixels changing state every 5 seconds (via mqtt) for over 1hour with no resets or other issues.

Great. Would you mind uploading your sketch somewhere, e.g., on https://gist.github.com/?

Makuna, 2016-02-04:
I have updated the UartBranch again with even more sample fixes.

closed 2016-02-23 by Makuna