Makuna/NeoPixelBus #30: void Lighten(uint8_t delta) loses colour information

Richard210363, 2015-10-03:
The code below moves the brightness of an NeoPixel that is set to Blue up and down.

Darken is fine.  However, when I use the Lighten command I can see the other Leds in the NeoPixel turn on and I have white light.

However, when the NeoPixel  darkens again the low level light is blue.

What is Lighten actually doing to the NeoPixel?

Makuna, 2015-10-03:
Are you running your esp at 80mhz?  There is an issue where the UART version must run at 160mhz.  This is noted in the Readme.

You can set this in the boards menu.  There are been no effects noted when running at 160mhz rather than 80mhz.

Let me know if that fixes your issue.

Since you have them working with the other branch, then you are either using level shifters for the IO or you are powering your pixels with near 3.8v.  If you are not, note that the pixels require the data line to be within +-0.5v of their Vcc to run reliably.

Richard210363, 2015-10-04:
Thanks for the quick reply.

I suspected that Lighten and Darken must be doing something like that.

I'll change how I blend colour.

You may consider changing the readme, as it says "this will lighten the color by the given amount".  You see "lighten" as blend to white, I read it as "increase the intensity of the light".  
Semantics :-)

Richard

Makuna, 2015-10-04:
Will do.

Makuna, 2016-01-13:
I updated master, UartDriven, and DmaDriven branches to be more specific in the readme

closed 2016-01-13 by Makuna