Makuna/NeoPixelBus #33: Stable with ESP8266

hallard, 2015-12-02:
@Makuna ,

I'm using this lib on all my ESP8266 boards (thanks for it, it rocks!). 
I just noticed that you recommend using UartDriven version at 160MHz but on my boards I just have 1 LED for visual information. I'm running at 80MHz without any problem on GPIO0 (so not UART driven) with SDK 1.3.0 (in fact arduino ESP8266 branch) and of course Wifi activated.

Do you have any more information of the crash with Wifi problem because I don't see it ? May be it's related with number of LED driven ? Do you think it's safe to drive 1 LED without Uartdriven method ?

thanks for any explanation.

Makuna, 2015-12-03:
The issue with the UART version needing 160mhz is something to do with the timing of shoving the bits through the hardware UART.

If you are only driving One LED you are unlikely to hit any problems.  The issues are seen more with longer strings of LEDs.  You may still see it once in a while but it will be rare with only one.  Also, if you update the LED often (a few times a second or faster), one update may get corrupted but the next will fix it so you may not see it.

hallard, 2015-12-03:
Thanks for this clarification, I blink the led once a second ;) except in setup where it's blinking very fast during different phase such as wifi connection, ota wait, Init, ....

closed 2015-12-29 by Makuna