Makuna/NeoPixelBus #35: Lights not being set correctly using uart branch

bnolan, 2015-12-14:
Hey all, great library. I am powering about 200 neopixels off a nodemcu 1.0. I managed to get the uart branch to work by hooking up the neopixels to GPIO0 and a level shifter, but it doesn't appear very reliable. I set the code to flash the entire strip on and off with green, but I keep getting flashes of other colors appearing, which makes me think that the timing is not working properly.

Has anyone else had this problem, is there anything else I can try? I am recieving OSC signals on the esp8266 and using it to sequence our lighting rig, but at this stage it looks like we're going to have to go back to using a raspberry pi.

Makuna, 2015-12-14:
Are you running the esp at 80mhz?  Try 160mhz.  
Are you running the latest Esp Arduino builds?
Confirm the pin, I thought it was GPIO2 for the uart.

bnolan, 2015-12-14:
I had it running at 160mhz, didn'try try 80. I'm pretty sure I had the right pin because it was sometimes doing what I sent it, but other times it'd just get a bit unreliable. Thanks for your help.

Makuna, 2015-12-29:
I updated the uartdriven branch with some changes.  Could you give it a try to see if it solves your problems.

alltheblinkythings, 2015-12-30:
I saw your changes in da00d519138f380c622f9888f58a86aa8c929654.  I haven't tried because I am away from my hardware right now but when I tried something similar to this it didn't work out for me - I still had substantial visible flicker :-( .

I'd been separately looking into the problems with the UartDriven branch and spent some time splitting out and proposing some fixes for the mainline serial code over at https://github.com/esp8266/Arduino/pull/1320.  With that change plus an `InterruptLock` held over just the `Serial1.write()`, I didn't see any visible flicker when looking at a strand of 60 LEDs hooked up to my device and running at 80MHz.  I still wanted to hook up to a logic analyser and see whether we'd benefit from disabling interrupts around all three bytes of a pixel rather than just each colour, as well as confirming things work with some longer strands.  Would love to hear if this helps you at all.

Some notes if you want to continue down the path of writing directly (which I haven't tested, just observations from working on the HardwareSerial code):  I think you don't need to wait as long as you do in the UartSendByte routine. The FIFO doesn't need to be completely empty, just empty enough for the next byte to be sent.  You also might be able to optimise further by only waiting for room in the FIFO once and then stuffing all the bytes for each pixel in in a burst.

Makuna, 2015-12-30:
@alltheblinkythings locking the irq will not work.  It will cause resets if you run any wifi code.  If this wasn't a problem then my main branch would work, but alas it doesnt.  

I tried only waiting before I send the four bytes but this created inconsistent timing in the output.  I really wish the sdk wasn't so temperamental with irq locking.

alltheblinkythings, 2015-12-30:
Disabling IRQs for a very short period of time (such as while you stuff bytes into a buffer) should be perfectly fine.  The SDK programming guide (2C-ESP8266__SDK__Programming Guide__EN_v1.5.pdf) explicitly states it's OK to disable them for 10us.  I think each of the four bytes takes 2.5us worst case to enqueue (8 bits at 3.2MHz), so we should be OK per that rule, even if the FIFO is full.

HardwareSerial had a deadlock bug (causing a WDT reset) if you called write() with interrupts disabled and caused its internal circular buffer to fill, and that lockup is fixed in my pull request.  On top of that, there were resets if other SDK code squeezed the HardwareSerial code out of the RAM cache of flash code.  Perhaps these were related to the failures you saw?  (This is why I said I'd love to hear if these patches worked for you - I haven't experienced any device level resets and WiFi still works after over 20h of continuous runtime with my changes.  Regardless of whether you try it, we should still fix the HardwareSerial bugs, though.)

I wonder why you observed inconsistent timing with the output in the case of stuffing multiple bytes at once - seems like allowing the FIFO to run dry would cause more jitter.  Any theories?

Makuna, 2015-12-30:
@alltheblinkythings ok, so what I was seeing was not what I expected, so I could only wait after filling the fifo buffer, but I had to make sure the send code was cached into ram.  I have updated the UartDriven to make sure the function is placed in RAM for consistent timing of the running statements.

I went with waiting every four uart bytes otherwise I include another statement to check if fifo is filled.  This is more important when running at 80mhz were the code in the loop is close to the actual timing.

alltheblinkythings, 2015-12-30:
Ah, I missed that your branch didn't have this forced into RAM.  That might
explain some of the weird timing you saw.  I think SDK1.5 trashes the IRAM
cache a lot more.

I'll try this tomorrow when I'm back home, but I am almost certain this
still won't work on at least one of my modules (ESP-202) due to there being
very little tolerance.  Not sure why but the 202 seems to have different
performance from NodeMCU device when WiFi is on.  If an interrupt comes in
between the writes then the timing will still be bad.

The best behaviour I've experienced when not using the HardwareSerial code
has been from:
  - build 12*N bytes of data in buff (i.e. whole pixel)
  - poll for tx fifo less than (128-12*N) bytes of data
  - InterruptLock (note: don't use noInterrupts()/interrupts() because
you'll turn interrupts back on if they were off at entry)
  - dump all 12*N bytes into FIFO
  - implicit interrupt enable (when interrupt lock goes out of scope)
  - make sure the end time recorded accounts for the bytes still in the tx
fifo

This gives us a good burst of data to tide us through any interrupt that
arrives immediately after interrupts re-enable, building up as we go
through more loops, when we quickly hit a full TX fifo (128*2.5us worth of
buffered data can handle quite a few interrupts).

With HardwareSerial I had no glitches with a "cork" routine that deferred
sending until when the internal cbuf was full, which would ultimately kick
off 128 bytes of writes from the tx empty interrupt handler (i.e. with
interrupts disabled).  This is not much different from what I described
above but without the HardwareSerial internal buffer to absorb larger
bursts of interrupts.

Makuna, 2015-12-31:
@alltheblinkythings lets discuss solutions in the gitter chat.  There is ton I have learned since I created the first version (like this annoying level 14 IRQ in the wifi core that can't be masked any more).

Makuna, 2015-12-31:
Another update, please get the latest.
If you still see an issue, try the new experimental flag NEO_IRQLOCK and see if this fixes your issue.

Makuna, 2016-02-04:
I have updated the UartBranch again with even more sample fixes.

closed 2016-02-23 by Makuna