Makuna/NeoPixelBus #38: Uart branch flickers and flahes with 150 neopixels

weboide, 2015-12-29:
Doing a simple animation with 150 neopixels lighting up one neopixel after the other (code below), part of the strand flashes randomly.
I've got the Huzzah ESP8266 set up at 160Mhz in the Arduino IDE with the Uart branch of NeoPixelBus and the Esp Arduino library version 2.0.0. (Arduino 1.6.5)

```Arduino
#include <NeoPixelBus.h>

#define pixelCount 150 // this example assumes 4 pixels, making it smaller will cause a failure
#define pixelPin 2  // make sure to set this to the correct pin

#define colorSaturation 128

NeoPixelBus strip = NeoPixelBus(pixelCount, pixelPin);

RgbColor red = RgbColor(colorSaturation, 0, 0);
RgbColor green = RgbColor(0, colorSaturation, 0);
RgbColor blue = RgbColor(0, 0, colorSaturation);
RgbColor white = RgbColor(colorSaturation);
RgbColor black = RgbColor(0);
    
void setup()
{
    // this resets all the neopixels to an off state
    strip.Begin();
    strip.Show();
}

void wipe(RgbColor col)
{
    for (uint8_t pixel = 0; pixel < pixelCount; pixel++)
    {
      strip.SetPixelColor(pixel, col);
      strip.Show();
      delay(31); // 31ms ~= 30hz?
    }
}

void loop()
{
    wipe(red);
    wipe(black);
    
    wipe(green);
    wipe(black);
    
    wipe(blue);
    wipe(black);
    
    wipe(white);
    wipe(black);
}
```

Makuna, 2015-12-29:
Could you experiment and see if can find at what count it does consistently work?  49?

weboide, 2015-12-29:
48 is when it starts flashing, before that pixels #1 through #pixelCount are lighting up correctly and seem stable (although some pixels beyond #pixelCount are lighting up and stay on for some reason, probably due to no reset being sent to them).

Makuna, 2015-12-29:
I submitted some changes to the uartdriven branch, could you try these.  Let me know the results.  Again, try the 49 and then try the full count.

weboide, 2015-12-29:
Nice Job! the flashes stopped! I tried with 49 and with 150 and even decreased my delay to 1 (instead of 31) and things are smooth!

Makuna, 2015-12-29:
fixed with pull #39

closed 2015-12-29 by Makuna