Makuna/NeoPixelBus #47: ws2811 blink randomly

far5893, 2016-01-22:
ws2811 blink randomly ,see video
https://goo.gl/photos/QYTVuNe7evucHtU38

Can you add a metod to stop spi dma?
And suggest hardware modification to better ws2812 compliant.

Makuna, 2016-01-22:
Please provide more information.

This is the Dma branch?
How do you connect them to the esp?  
What speed are you running the esp at?
What flags are you passing the constructor?  Sample code demonstrating the issue?

Great suggestion about adding the stop api, please create a seperate issue for that and I will do so.

far5893, 2016-01-22:
1)It is dma branch.
2) A wire of about 5cm no level conversion i suppose not full hardware related because standard branch without wifi ON worked well before  sdk changes ,may be  timing problem + wifi interaction.
3) 80Mhz default
No flag ,
#define pixelCount 40 
#define pixelPin 5 

#define colorSaturation 128

NeoPixelBus strip = NeoPixelBus(pixelCount, pixelPin);

Makuna, 2016-01-22:
Another thing to try is the UartDriven branch.  Does it also not work?  It will require a different pin be connected.  Both the DmaDriven and UartDriven have been Wifi safe so far.

But what your video shows is what I see when I run the pixels at 5v and drive the IO at 3.3v (esp8266 is 3.3v).

Your hardware setup is more than likely the issue here.  Ws2811 requires that the digital IO line be within +-0.5v of the Vcc.  They also require the minimum Vcc to be 3.7v.  Outside these specs things are pretty inconsistent usually due to how much power is being drawn.  
Another thing is to make sure you put a large cap on the Vcc to the pixel strip (see Adafruit's guide to connections).  
My experimentation has found that 3.8v is really required to drive the Ws2811 and keep the digitial IO coming from the esp8266 (which is 3.3v) happy and running well. 
I recommend that you actually put in a level shifter for the digital IO to shift it from 3.3v to 5v and drive the ws2811 at the full 5v if you have more than one pixel.

Makuna, 2016-02-04:
I have updated the DmaBranch and it will no longer continuously send data by default.  
It also got some sample fixes so make sure any tests that were based on the samples get updated.

Makuna, 2016-02-24:
Is this still an issue?  I have many people that have this working fine.

closed 2016-03-01 by Makuna