Makuna/NeoPixelBus #50: Please add to Arduino Library Manager

probonopd, 2016-01-24:
Please make this library available in the Arduino [Library Manager](https://www.arduino.cc/en/Guide/Libraries#toc3). 

* Ensure your library is complaint with [1.5 format](https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5:-Library-specification)(1.5 format folder layout is not required)
* Tag it and push the tag, or create a release with github "releases"
* Open an issue on [Arduino's github](https://github.com/arduino/Arduino/issues), specifying the git repo (or github url) from where to download your library

More information on https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

Makuna, 2016-01-26:
working on it, but do they support branches at all?

probonopd, 2016-01-26:
Not sure, and I'm also not sure whether having different branches over a prolonged time is a good idea: The current situation is not ideal where users can have only one of the branches installed at a time and need always to take care to have the matching one for a given sketch. So possibly it would make sense to have different libraries under different names (e.g., "NeoPixelBusAnimator"), or to merge all the different modes of operation into one branch and let the user decide which .h files to include.

Makuna, 2016-01-26:
Well, esp01 can't support the dma model due to not exposing the pin, so I need two models for esp8266.  The solution makes the library more cryptic to consume without adding more branching code (templates) That's one issue.
the second issue is that avr arch is missing headers that are included in other arch that I require for my animations, so I have the older version for those.  The Arduino got less fragmented with 1.6 update but is still pretty annoying to support one library that does it all.

Makuna, 2016-02-24:
V2beta branch has one present but it is not yet registered. This branch will become the source in the near future.

Makuna, 2016-02-29:
As soon as the Arduino Library manager scrubber runs, it should show up.

Makuna, 2016-03-04:
Well, it should be available but there is a bug in the IDE that MAY not pick up the latest libraries.

If you don't see it, you maybe hitting this Arduino IDE issue: [#4139](https://github.com/arduino/Arduino/issues/4139#issuecomment-156857120)

The issue has be resolved, however they have not released a new IDE version with the fix [#4361(https://github.com/arduino/Arduino/pull/4361).

Suggested workarounds for now:
•Delete  %LOCALAPPDATA%\Arduino15\library_index.json.tmp.gz  and then restart the IDE.
•Or try an hourly IDE build.

Makuna, 2016-03-09:
Latest version of IDE with this fixed should be out. Closing this.

closed 2016-03-09 by Makuna

probonopd, 2016-03-09:
Great, thanks.

probonopd, 2016-04-02:
> Not sure, and I'm also not sure whether having different branches over a prolonged time is a good idea: The current situation is not ideal where users can have only one of the branches installed at [...]

Another user possibly fell into the same trap:
https://github.com/probonopd/ESP8266HueEmulator/issues/18