Makuna/NeoPixelBus #55: DMA driven not working

mumin50, 2016-02-01:
Hi i tried on 2 esp12e bords DMA driven library.
All leds are getting white... right amount of them but all white

Master works ok - when hooking up osciloscope i can see all 0 and 1 
But when do the same for for DMA driven i got only 11111111. and that causes all led to be white.
Frequency is ok. Also amount of data is ok. only data itself is wrong

Makuna, 2016-02-01:
Please list the board you are using, the model of pixels, and how you wired it.
Sample code is helpful also.

Please try the prototype branch DmasingleFire also.

mumin50, 2016-02-04:
Board esp8266 esp12e tried on 2 models:
1)http://nl.aliexpress.com/item/Free-shipping-ESP8266-ESP-12-serial-WIFI-Industrial-stable-version-A-full-test-board-Full-IO/32259646325.html?spm=2114.010208.3.158.BBsGbV&ws_ab_test=searchweb201556_1,searchweb201644_1_505_506_503_504_301_502_10014_10001_10002_10016_10017_10010_10005_10011_10006_10003_10004_401_10009_10008,searchweb201560_3,searchweb1451318400_-1,searchweb1451318411_6449&btsid=c58578ad-2751-4557-af8e-dbb6d80e3b4a

2)http://nl.aliexpress.com/item/Free-Shipping-50Set-ESP8266-WiFi-Module-Breakout-Board-Adapter-Plate-for-ESP-07-ESP-08-ESP/32421508298.html

Sample code - is an example from library (tried both) 

RDX0/GPIO3 witch one it is to make it clear i use RX pin (if antena on top 2nd from top left)

mumin50, 2016-02-04:
I think example is NeoPixelFun example is not working right ... i just used NeoPixelTest and i got 4 LEDs like it should  (?!) :)

mumin50, 2016-02-04:
IMO someone could check example and You can close this ticket

Makuna, 2016-02-04:
Are you running the sample without change?  (it has four routines in it that can be run, by uncommenting   one at a time.  I am curious if the others work.  I see nothing wrong with it with the number of pixels you are using (there is a bug if you have more than 255 pixels).

Please review [AdaFruits Neopixel Best Practices](https://learn.adafruit.com/adafruit-neopixel-uberguide/best-practices)

CAUTION:  Your setup for that ring is incorrect!  Please look at Adafruits links carefully, you are missing a large cap on the power in and a resistor.  The specific sample you are having an issue with is animating basically White (all LEDs on), which draws a lot of power in spikes, which without the capacitor can cause the internal cpu on each neopixel to mess up and do like what you are seeing.  Be careful with those rings, I have toasted two before I learned that they are also sensitive to power being attached well, too many spike burn out the first pixel. 

It also looks like you are connecting the ring to 3.3v?  Or is it 5v?  I don't see level shifters for the IO.  This is also a problem.  The pixels don't like running below 3.8v, and if run at 5v they don't like their IO line being more than +-0.5v from the driving voltage.  The Esp runs at 3.3v.  So either you run the pixels at 3.8v (variable power supply is the only way I know to get this) or you have to use level shifters.  Without fixing this you will not get a stable setup.

And this ring can draw a full 1amp.  The Esp boards can't source that easily, most are only rated at 750mA max.  While probably not your issue as you are only driving four, its just something you should be careful with.

Makuna, 2016-02-04:
I dug some more and did find a bug in the example.

Change the line...
```
// progress will start at 0.0 and end at 1.0 
RgbColor updatedColor = RgbColor::LinearBlend(originalColor, color, (uint8_t)(255 * progress)); 
```
to this...
```
// progress will start at 0.0 and end at 1.0 
RgbColor updatedColor = RgbColor::LinearBlend(originalColor, color, progress); 
```

Makuna, 2016-02-04:
I have updated the DmaDriven with the example fix.  Please let me know if this is fixed now for you.

mumin50, 2016-02-08:
Yes it is connected quick and dirty :) it was just proof of concept and it works great 

I tested new example... Yes it works but i miss "LoopFade" if I try to use it is not existing :)

But now im happy i using it in connection to Art-Net to drive ws2812b with DMX software.
And DmaDriven is not blocking WiFi...

It is just little slow. But i need to investigate it more because i'm not sure if this is because of Your library or whet is more possible it is wifi problem.

Thanks again for good work.

Makuna, 2016-02-08:
It got renamed to LoopAround at some point in the far past.    I will check history as to why the name got changed and I noticed that it was not changed in one part of the example.

Makuna, 2016-02-24:
V2beta branch is becoming the one and only source going forward, and all these fixes were applied there also.

closed 2016-02-24 by Makuna