Makuna/NeoPixelBus #58: Minor warning to correct.

far5893, 2016-02-02:
I have some wanings in my eclipse IDE , i don't know if they are reported in Arduino IDE.

Description	Resource	Path	Location	Type
Member 'H' was not initialized in this constructor 	HslColor.h	/esp8266_rgbwebstrip/Libraries/NeoPixelBus/src	line 56	Code Analysis Problem
Member 'L' was not initialized in this constructor 	HslColor.h	/esp8266_rgbwebstrip/Libraries/NeoPixelBus/src	line 56	Code Analysis Problem
Member 'S' was not initialized in this constructor 	HslColor.h	/esp8266_rgbwebstrip/Libraries/NeoPixelBus/src	line 56	Code Analysis Problem
Suggested parenthesis around expression 'PIN_FUNC_SELECT(PERIPHS_IO_MUX_U0RXD_U, FUNC_I2SO_DATA)'	NeoPixelBus.cpp	/esp8266_rgbwebstrip/Libraries/NeoPixelBus/src	line 191	Code Analysis Problem
Member 'B' was not initialized in this constructor 	RgbColor.h	/esp8266_rgbwebstrip/Libraries/NeoPixelBus/src	line 62	Code Analysis Problem
Member 'G' was not initialized in this constructor 	RgbColor.h	/esp8266_rgbwebstrip/Libraries/NeoPixelBus/src	line 62	Code Analysis Problem
Member 'R' was not initialized in this constructor 	RgbColor.h	/esp8266_rgbwebstrip/Libraries/NeoPixelBus/src	line 62	Code Analysis Problem

Makuna, 2016-02-02:
The warning about PIN_FUNC_SELECT is due to the macro in the esp8266\Arduino project.  It is nothing I should fix in my project.  Please create a issue in that project.

The other warnings are by design.  The constructor in question has a comment cautioning the user about not initializing the members themselves.  Due to this library needing to work on very memory restricted devices, there is a need to have such a constructor that doesn't take the time/code memory to zero init them and then have the calling code over write them with the values latter.  I will see if they can be suppressed with a pragma.

Makuna, 2016-02-02:
Ok, After some thought, it is good that the warning is displayed.  These should be only used in very specific cases and the user should know why they use them and the issue, this extra warning is good to raise the issue so they can make a good decision based on their code.

closed 2016-02-02 by Makuna