Makuna/NeoPixelBus #68: Only bitbang works on my esp8266

wunderbaum, 2016-02-28:
I tried all methods to get my stripe working. But only the bitbang-method works (so this shows me that I am using the right gpio-pin). With DMA, nothing happens. I am using an esp12e and gpio pin 4.
What am I doing wrong?

Makuna, 2016-02-28:
Its the Pin.  Dma and Uart have hardware limited pin and the passed in Pin value is ignored.

Look at the page [Wiki On NeoPixelBus](https://github.com/Makuna/NeoPixelBus/wiki/NeoPixelBus-object) and specifically the NeoDma800KbpsMethod, DMA for the old and new libraries only works on a single pin no matter what you pass in.  Its a limitation of the hardware.  That pin is GPIO3, pin 3.
Uart version has a similar limitation but its another pin, GPIO2.

What branch are you using?  I suggest you use V2Beta as it is about to go live as THE ONLY version.  It will replace master and become available in the Library Manager when it goes live.

wunderbaum, 2016-02-28:
Ah!
I am using V2Beta, yes. is not easy to find out the most actual version as there are so many forks.
Thanks alot for your support. That I have not clearly understood while testing the examples. I tried all gpio except the tx/rx ones and was totally confused as I tried with the bitbang and it worked.
In the examples its not mentioned that dma is fixed to a gpio. 

I can sleep well now, many thanks! I owe you a beer!

Date: Sun, 28 Feb 2016 10:35:50 -0800

Makuna, 2016-02-28:
Good feedback, I will update the samples with a few comments.  Also feel free to use the chat channel on gitter as others maybe able to help you faster.

https://gitter.im/Makuna/NeoPixelBus

closed 2016-03-01 by Makuna