Makuna/NeoPixelBus #74: HsbColor with Hue set to 1.0f

plutots, 2016-03-01:
If I use HsbColor(1.0f,1.0f,1.0f) the pixel seems not do be correct Red. There is a big jump between Hue 0.999 and 1.0f. Just checked the Hsb to Rgb conversation and there is no case to handle hue values of 1.0f. I am using Esp8266 UART.

Makuna, 2016-03-01:
I am not seeing what you are describing.  The conversion is tolerant around the edges.  
here is the values of a hsbColor converted to rgb, then that rgb converted to hsl, then rgb converted back to hsb.  Note that 1.0 wraps to 0.0 as it should.

Hue Near Zero
```
hsb(-0.01,1.00,1.00) -> rgb(255,0,15), -> hsl(0.99,1.00,0.50), -> hsb(0.99,1.00,1.00)
hsb(0.00,1.00,1.00) -> rgb(255,0,0), -> hsl(0.00,1.00,0.50), -> hsb(0.00,1.00,1.00)
hsb(0.01,1.00,1.00) -> rgb(255,15,0), -> hsl(0.01,1.00,0.50), -> hsb(0.01,1.00,1.00)
```

Hue Near 1.0
```
hsb(0.99,1.00,1.00) -> rgb(255,0,15), -> hsl(0.99,1.00,0.50), -> hsb(0.99,1.00,1.00)
hsb(1.00,1.00,1.00) -> rgb(255,0,0), -> hsl(0.00,1.00,0.50), -> hsb(0.00,1.00,1.00)
hsb(1.01,1.00,1.00) -> rgb(255,15,0), -> hsl(0.01,1.00,0.50), -> hsb(0.01,1.00,1.00)

plutots, 2016-03-01:
if i do something like that:
```
RgbColor rgbC;
pixels.SetPixelColor(0, HsbColor(0.99f,1.0f,1.0f))
rgbC = pixels.GetPixelColor(0);
serial.println(rgbC.R);
serial.println(rgbC.G);
serial.println(rgbC.B);
```
I get
> 255
> 0
> 15

and with this:
```
RgbColor rgbC;
pixels.SetPixelColor(0, HsbColor(1.0f,1.0f,1.0f))
rgbC = pixels.GetPixelColor(0);
serial.println(rgbC.R);
serial.println(rgbC.G);
serial.println(rgbC.B);
```

> 255
> 0
> 255

but that should be 

> 255
> 0
> 255

Also the LED looks bright pink instead of Red.

Makuna, 2016-03-02:
Thanks, your pull was on the right track, but I want to avoid expensive calls like floor().  I comment why in your pull.

closed 2016-03-02 by Makuna

Makuna, 2016-03-02:
thank you for your persistence on the issue.  My test which increments by 0.001f from -0.9f to 1.1f didn't trigger this issue as some "floating point" magic didn't happen when it hit 1.0f.  But when I put a specific 1.0f into the hue it triggers.  I added this to the test.

plutots, 2016-03-02:
I just tried the new version, works as it should, nice!