Makuna/NeoPixelBus #78: ESP8266 different Mode

hallard, 2016-03-02:
I'm designing a new PCB board and referring to my old issue #33, I have two questions :
- does the bit bang mode still has problem with WiFi enabled or is it solved with new SDK (1.5.1) ?
- I think I know the answer, but just in case, using UART Driven or DMA mode remove me the ability to use UART as normal serial for other tasks, right ?

Thanks for any help

Makuna, 2016-03-02:
I have not tested with the latest Arduino release yet to see if BitBang works with WiFi active.

But, I don't think BitBang will ever be functional with WiFi active.  I haven't heard a change in their belief on issue; they have stated they don't plan to solve the problem; and they suggest hardware solutions.  I continue to include it in the hopes some day they may change their minds. 

UART method will stop the secondary Uart from being usable, but the default "Serial" is still fully functional as it uses a different one.  The Esp has more than one. 
DMA method will stop the Serial read, but not the Serial write.  It uses the UART0 RX to as output, but the UART0 TX is still available.  So doing serial debug output is still functional, you just loose the ability to read from it.  There is also no problem flashing with Pixels connected.

In either case, if you need another UART, there is one available.

hallard, 2016-03-02:
@Makuna 
thanks for this answer.
When you write about 2nd UART, are you refering to the one used for Debug GPIO2/TXD1 ? So UART Mode use this one or is it another one I don't know ?

Thanks

Makuna, 2016-03-03:
Rather than me getting it wrong, let me point you to documentation ;-)

About half way in is the Serial section.
http://esp8266.github.io/Arduino/versions/2.1.0/doc/reference.html

> 
Serial uses UART0, which is mapped to pins GPIO1 (TX) and GPIO3 (RX). Serial may be remapped to GPIO15 (TX) and GPIO13 (RX) by calling Serial.swap() after Serial.begin. Calling swap again maps UART0 back to GPIO1 and GPIO3.
Serial1 uses UART1, TX pin is GPIO2. UART1 can not be used to receive data because normally it's RX pin is occupied for flash chip connection. To use Serial1, call Serial1.begin(baudrate).

You can see that UART0 RX is GPIO3, the same as DMAs Output.  
You can also see that "Uart Method" uses GPIO2, which is the Serial1 TX pin so it doesn't effect flashing or Serial output.

The Esp8266 allows for remapping of the pins for hardware features, but only a limited set for each hardware feature.  Lots of overlapping, pretty much only leaving the pins that I documented for Uart Method and Dma Method and should not cause undue issues with normal use.
Clear as mud ;-)

hallard, 2016-03-03:
@makuna,
Thanks that was exactly the sense of my first question, I just wanted to be sure I didn't missed anything ;-)
So in case of other users interested here what I understood, just correct me if you think I'm wrong
* Using DMA mode, take the UART0 Serial RX (GPIO3), this is why it can be not so clear because for this pin, you could think it's input, but DMA use it as output. The good point is that this let you use TX (GPIO1) to send some debug info using classic `Serial.begin() bla bla`  May be caution should be taken on init order if you want to use `Serial.print()` because calling `Serial.begin()` would reconfigure RX as input and possible trouble DMA or vice-versa.
* Using UART driven use GPIO2 so TXD1 (`Serial1.begin()` in code) so you will loose this "system debug" UART1 but may me the best solution because in this case you keep all "classic" serial untouched ;-)

Makuna, 2016-03-03:
Confirmed that with DMA method, setting Serial.being() after the strip.Begin() will cause the strip to not function correctly.  I will capture all this in the wiki.

Makuna, 2016-03-04:
Captured in Wiki

closed 2016-03-04 by Makuna