Makuna/NeoPixelBus #80: how to declare extern bo be usuable in all files

hallard, 2016-03-04:
Hi there,

I've got lot's of .cpp files in my project, each with one .cpp and one .h file. Before V2 I was declaring rgb let strip as follow in my `main.ino` file

```arduino
  ESP8266WebServer server(80);
  NeoPixelBus rgb_led = NeoPixelBus(1, RGB_LED_PIN, NEO_RGB | NEO_KHZ800);
```

And in main.h as follow
```
  extern ESP8266WebServer server;
  extern NeoPixelBus<NeoRgbFeature, NeoEsp8266BitBang800KbpsMethod> rgb_led();
#endif
```

All was working fine but with this new version, I'm sorry I'm not a cpp fan and I don't know how to do this, I tried doing the same but looks like extern template works different

```arduino
  ESP8266WebServer server(80);
  NeoPixelBus<NeoRgbFeature, NeoEsp8266BitBang800KbpsMethod> rgb_led(1, RGB_LED_PIN);
```
And in `main.h` as follow
```
  extern ESP8266WebServer server;
  extern NeoPixelBus<NeoRgbFeature, NeoEsp8266BitBang800KbpsMethod> rgb_led();
```
but it fire a
```
error: 'NeoPixelBus<NeoRgbFeature, NeoEsp8266BitBangMethodBase<NeoEsp8266BitBangSpeed800Kbps> > rgb_led' redeclared as different kind of symbol

   NeoPixelBus<NeoRgbFeature, NeoEsp8266BitBang800KbpsMethod> rgb_led(1, RGB_LED_PIN);

```
any idea how to achieve this ?

Thanks

Makuna, 2016-03-04:
If you search online this is common problem to hit, and sometimes its hard to see the tree for the forest.
Your problem is in this code...
```
 extern ESP8266WebServer server;
  extern NeoPixelBus<NeoRgbFeature, NeoEsp8266BitBang800KbpsMethod> rgb_led();
```
One line is correct, one is not.  You have () on the rgb_led so the compiler thinks its a function rather than a class.  remove the ().
It should look like...
```
 extern ESP8266WebServer server;
  extern NeoPixelBus<NeoRgbFeature, NeoEsp8266BitBang800KbpsMethod> rgb_led;
```

I would also suggest that you typedef the NeoPixelBus so that if you want to change it, you change it in one spot.  Further makes it easier to read code.  So in main.h
```
typedef NeoPixelBus<NeoRgbFeature, NeoEsp8266BitBang800KbpsMethod> MyPixelBus;

extern MyPixelBus rgb_led;
```

hallard, 2016-03-04:
Works like a charm, thanks, and yes typedef is much better ;-)

closed 2016-03-04 by Makuna