Makuna/NeoPixelBus #81: please add more example like the fastLED library

mkeyno, 2016-03-05:
hi @Makuna , thanks for increasing the example but I think the rich example such as FastLED library would be so great

Makuna, 2016-03-05:
Do you have a link to this library?

I tend to make examples from common techniques to demonstrate best practice.  Do you have a specific request?

mkeyno, 2016-03-05:
hi @Makuna  , thanks for quick reply but I'm supersized you never check [FastLED library example](https://github.com/FastLED/FastLED/tree/master/examples) , many people  use its concept for lighting project but not totally adapted for extended demanding ESP8266 module, I have lot of fun with this library but like others guys I'm moving to ESP core but unfortunately  not can what every I did with AVR module, however may people quote your library but the example not rich as FastLED is , and I hop it will be extended , because I've tried to move from Teency POV  to ESP POV

BR

Makuna, 2016-03-05:
Most of those examples can port over easily.  A lot of them are matrix panel specific which doesn't represent all customers.  

Are you really looking for a library of effects and not just examples?

mkeyno, 2016-03-05:
frankly FastLED give us the tools like palette design so I could care only the effect manipulation however in ESP I should care about handling the client so running the effect when you have to care other stuff would be challenging , that's why I believe you library should have sample for that  for example ,  I've try to make webserver when we could select different effect or bind the special effect to the sensor  event, but making the effect is so time consuming and hard for me regards the really fast FastLED library

Makuna, 2016-03-06:
Lets move this conversation over to gitter channel for so I can understand more of your needs.

mkeyno, 2016-04-12:
dear @Makuna I used your last example in my ESP module and it is worked very well ,and thank you very much for add those new example and sharing it however , normally we use ESP as weserver or clientserver,  can you  make couple of example whereas have both task together in loop, for example check the new client request while show the effect  , and in new client request we change the show effect

Makuna, 2016-04-12:
This library is purely about NeoPixels across all Arduino boards, that example would be specific to Esp8266.  I want to refrain from that since the actual NeoPixelBus calls would similar to all the current examples.  The only real hint is to only ever call StartAnimation/StopAnimations or SetPixelColor from network callbacks, and merge in the two calls in `loop()` to `animator.UpdateAnimations()` and `strip.Show()` and it will just work.  
See [Melvanimate](https://github.com/sticilface/Melvanimate) for a project that does something similar to what you are looking for.  I have a link to this project; and soon more; that use my library in more advanced scenarios.

closed 2016-05-22 by Makuna