Makuna/NeoPixelBus #83: Examples are really hard to follow

NickWaterton, 2016-03-07:
I'm using an esp8266, and this library seems to do exactly what I want.

I actually wrote my own routines to do the same thing (using the UART), but they are hackish and I don't like them.

I've been trying to decipher your examples, but they are very hard to follow. All the random colours and effects don't help at all. Mostly my pixels come on white and twinkle pastel shades. I'm sure that it's working, but just a few real colours (red green blue) with fixed durations/effects would make it so much easier to read and understand.

An explanation of what an "animation channel" is, why would you use one for timing others? Why would I need more of them? For example.

Why use floats for intensity and progress? How does that work on a micro controller with no floating point support? (Software emulated I suppose -but why???) 0 to 254 is so much easier to understand, and why fit 255 steps into 100 (assuming 0.0 to 1.0 is 100 steps). Sorry, it's just baffling me.

Thanks for your time and effort, I'm just having a hard time getting it. Once I've got the concept, I'm sure it'll all come clear.

Makuna, 2016-03-07:
I am constantly improving the docs and examples.  They get better every day due to good feedback.  The examples start simple and get more complex on purpose.  There is always the Gitter channel to ask others who are using this library. If you have suggestions for examples, please provide a specific idea.  

I will try to answer some of your questions here. But alot of this follows standard animation techniques. I would suggest anyone getting into animating do a little research on how animations work in software in general and not juse one implementation for a specific case someone has . 

This contaims an animation framrwork, it doesn't provide everything everyone wants, it provides a flexible core for users to build from.  The most common pieces are here, and it won't fit every need as is. 

The  animation channel is a async tracker of progress through time.  You need one for every effect that is running but that also needs a different time period.  You may have one channel turning one pixel on that takes 5 seconds, and another dimming another 15 pixels over one second.  You may have yet another moving a lit pixel across part of the strip.  What the channel actually accomplishes is up to you, it just tracks the time and provides a progress over the defined time in an async fashion.  No more using delay().

Progress is a float for many reasons.  Top reason is that there are plenty of routines that fit within my model out on the internet, almost all are coded for floats with 0.0 to 1.0 values, so the average Arduino user could find and just use them.

The duration of the animation is variable over very short or long periods, having an abstraction allows for better design of effects by enforcing seperation of time and the actual effects, without the hard rquiremnts lIke only 100 steps, or even 256.  It's a float, the increment is arbitrary based on duration and how fast your update loop is .  Having an animation that runs over an hour only having 256 steps would be very stuttery.  Remember animation isn't just color change, so assumimg only needimg a rgb resolution of 256 is a misstake.

To better provide physics feel of acceleration to the animations , you will need to modify the progress also in an easy abstracted way.    Thus the standard  easing functions (I didn't invent those, it's standard animation techniques, and again 0.0 to 1.0 values are common). These functions are best written as floats woth0ut beING overy specific to a single use.   I am working on a Cylon red eye example that makes their use very clear and interesting.

Blending functions also often take a float for similar reasons and for some classes of effect, this is what you are really doing, blending.  Even movement is just interpolation between two locations over time.  Curves are two points, some controls, and a progress.   Finding a blend routine on the Internet that uses anything other than floats is difficult.  Again, I provided some color blends, but there are many cool routines you can just find and plug in.

I made this to be simple easy to maintain code library, smaller code, but it may not be the fastest. But it's also not overly specialized or limited. Design decision tradeoff.

I have no idea how many pixels you have in your design, but this works fine on small Avr arduino with 50 pixels doing a seperate animation per pixel.

diegozalez, 2018-08-02:
First of all thank you for the great library you make i just have a problem undestanding this example, im kind of new to animations i try other places to find info about animation in software but found nothing.
NeoPixelBus/examples/bitmaps/NeoPixelBitmap/NeoPixelBitmap.ino
Im using the lastest arduio IDE and the lastest version of this library, ESP32 DevKit, 140 NeoPixels in a 5*28 matrix. 
On this example i have some questions:
 1- const uint16_t AnimCount = 1;
     NeoPixelAnimator animations(AnimCount);
     Wat is the fuccion of this and when should it be change ?
2- const uint16_t PixelCount = 144;
    The pixel count is 144, im asuming the matrix have a total of 144 pixels but i later see the image on 
    the SD card is 144*144 for a total of 20736pixels 
3-And last of all some video of how this should work will be healpfull

closed 2020-12-13 by Makuna