Makuna/NeoPixelBus #84: Pixel length limit?

stedaniels, 2016-03-09:
Hi Michael,

I was wondering if there was a pixel length limit?  What's the longest length you'd been successful at driving with your NeoPixelBus sketch?

I'm hoping to drive two 300 pixel strips (joined together with power injected at multiple points).

An item in the FAQ with yours and other peoples success stories might be a good idea?  Once my Esp8266's arrive I'll submit a PR with my trials.

Cheers,

Steve

Makuna, 2016-03-09:
Maybe a whole section on customers scenarios?

I have looked at the data for 3000 with a logic analyser and it looked good.  On the esp8266 using the default dma method, the only issues are memory  (3000 x 3 x 4 = 360,000 bytes) and time to update that buffer.   While sending data stream is by hardware and thus takes little cpu overhead, it will limit the refresh times, as it takes 30us to send single pixel, 90ms to send 3000, that means about 11 updates per second.

stedaniels, 2016-03-09:
Thanks for the quick reply, I had no idea it would be that efficient.

Makuna, 2016-03-09:
I am going to leave this open as a reminder to improve the Wiki.

Thanks for your feedback.

Makuna, 2016-04-01:
Ok, with the latest DMA redesign, there is now a 341 pixel limit due to the single buffer I am using.  I will be addressing this; but I wanted to add a note incase someone comes across my comment above about 3000.

Makuna, 2016-04-04:
I have two new wiki seconts,
[How many pixels does this library support?](https://github.com/Makuna/NeoPixelBus/wiki/FAQ-%237) and
[Project References](https://github.com/Makuna/NeoPixelBus/wiki/Project-References)

closed 2016-04-04 by Makuna