Makuna/NeoPixelBus #85: allowing async operation / framebuffer mode

g3gg0, 2016-03-11:
hi makuna,

i see the code has developed very well. good work.
but what i am missing right now is the possibility to use the valuable CPU time for useful stuff.

i understand that there is some compatibility issue because it doesn't block just like the old bitbang code did.
but adding code to make it behave more like a framebuffer device would make it more flexible.
this will deliver you a standalone engine that just feeds RAM content to you beloved WS2812 without any CPU cost and
the CPU will do it's stuff, updating the RAM from time to time and handle network IO etc.

proposal:

public void NeoEsp8266DmaMethodBase:StartDmaAsync()
  configures the I2S buffer chain to:  data->latch->data ...

public void NeoEsp8266DmaMethodBase:WaitSync()
  waits (polling) for EOF that is only set when reaching the latch-buffer (it behaves like a vblank period, allowing code to fill buffers)

BR,
g3gg0

Makuna, 2016-03-11:
The support you want is already present for the Dma method.  Show() will copy data from back buffer to DMA buffer then trigger the DMA to start sending and return immediately.  It will only wait if the last dma send is still in progress, which means you were faster to prepare the next frame than it takes to actually send the data to the pixels.  You can externally test this by calling CanShow().
Does that make sense?

g3gg0, 2016-03-11:
aiiii, right. misunderstood the code.
all fine, thanks :)

Makuna, 2016-03-11:
Did you review the wiki?  
Was it discoverable?
Was it clear enough?

g3gg0, 2016-03-11:
after you wrote, i stumbled in the wiki and found it to be explained very well.
thanks for the credits btw :)

if i should suggest a change - it would be a section for the impatient ones like me.
more like a cheat sheet.

Makuna, 2016-03-11:
What sort of things would you like to see?  I am actively collecting info for the next pass at doc improvements

closed 2016-03-18 by Makuna