Makuna/NeoPixelBus #88: Not working on Wemos D1 Mini (ESP8266)

TotallyInformation, 2016-03-30:
Hi, there seems to be one or two issues when running on the D1 Mini.

Firstly, the data pin seems to be fixed on Arduino pin 3 which is the RXD pin on the D1 Mini and causes the whole string to go to max output when restarting.

Secondly most of the examples don't work. The Cylon example completes a single cycle, the others don't do anything though the ones that have serial output clearly show they are cycling.

This is a shame as I like the capabilities of the library.

I'm working with an 8 LED WS2812 5050 based neopixel. It works fine with the Adafruit library.

Makuna, 2016-03-30:
Did you read the Wiki on Esp8266 support?  Specifically [here](https://github.com/Makuna/NeoPixelBus/wiki/NeoPixelBus-object).  You may also want to review the [FAQ](https://github.com/Makuna/NeoPixelBus/wiki/FAQ)  

Due to hardware limitations, the default method (NeoEsp8266Dma800KbpsMethod) uses hardware that gives the user no choice of pin; but its the best mode for Esp8266.  
The Uart method (NeoEsp8266Uart800KbpsMethod) also has a limited pin and is the second best method (ESp01 are stuck with this, and it maybe the pin restricted Wemos D1 is also).  

The Adafruit library support for esp8266 is based on my work (credited in the code file) for the bitbang method (NeoEsp8266BitBang800KbpsMethod) and they both will have the same issue that if you use WiFi,  they won't work reliably.  This is due to the WiFi core will interrupt the data stream causing all sorts of glitches when WiFi is active, and you can't block these interrupts as doing so will throw an exception and reset the chip.   I keep it around for testing primarily.

CraigLindley, 2016-03-30:
I have a NodeMCU Amica and have been trying to run the demos and some code of my own and I am also having problems with stability. I have a 74AHCT125 level translator and a 100 ohm resistor between the Amica and the 8x8 Adafruit neopixel panel. The demos start to work and then stall completely; the same is true of the code I have written.

Makuna, 2016-03-30:
@CraigLindley Could you try the work around of passing a larger number to the constructor than you actually have but only reference the original count?  Add 50 to it.
(p.s.  I just ordered some WeMos D1 Minis, but it will take a bit to get here).

CraigLindley, 2016-03-31:
Adding 50 to the led count actually made the problem worse.

-- 
Craig Lindley / Heather Hubbard

New Recordings: craigandheather.net/cnmpage.html
Latest rock CD: craigandheather.net/hellinahandbasket.html
Latest non-rock CD: craigandheather.net/craigdoesfingerstyle.html

Personal Website: craigandheather.net

Phone: (719) 495-1873
Cell: (719) 502-7925

If you’re one in a million, there are now seven thousand people exactly
like you.

TotallyInformation, 2016-03-31:
Re the Docs/FAQ - yes I saw that. I think that the problem underlying this is that the ESP8266 platform is now much bigger. The Wemos D1 Mini is based on the ESP12EX model which is far more capable than the original ESP01 and is specifically set up to be very Arduino-like.

So perhaps the library should not be making choices itself or at least should allow them to be overwritten. I did try to look through the code to see if I could relax the restrictions but I'm afraid my C is rather limited.

I will try the other suggestion too. But if there is a way to not use the ESP code, that would be useful.

TotallyInformation, 2016-03-31:
Not sure if this might not shed some light on things: http://tech.scargill.net/esp8266-ws2812b-leds-on-a-plate/

Makuna, 2016-03-31:
@TotallyInformation Some of this is best talked about on gitter as to not add noise to the core issue here.  What feature exactly do you think the library is making a choice and not exposing a way to override?  

If you look at Wiki, it explains how to switch between different methods to use Dma, Uart, and even Bitbang.

Makuna, 2016-04-01:
Update:  I was curious why I wasn't seeing this when others obvious were, but not everyone.  So I followed my own often suggestion and used the "nuclear option" to clean Arduino and Esp8266 off my machine and started over with the latest.
AND I NOW SEE THE ISSUE!.  DMA is unusable and I am investigating.  So far what I have learned is that the DMA hardware freezes on a block won't automatically progress.

Makuna, 2016-04-01:
I just put in a redesign of the DMA method, underlying hardware use is similar but due to SDK change something I relied on no longer worked.  The new design should be future proof and should work on EX chips also (I don't have one to test with yet).

TotallyInformation, 2016-04-02:
Thanks, will try when I get a chance.

CraigLindley, 2016-04-05:
Seems like DMA on the NodeMCU Amica is working perfectly now. Thanks Makuna

hallard, 2016-04-06:
@TotallyInformation and @Makuna 
just to let you know, I played all day on my firmware working on D1 mini, I have 17 WS2812 connected to D0 (1 on board and 16 into a ring) using bitbanging, with Wifi, I2C, big webserver with bootstrap framework on SPIFFS and strip animation (breathing LED) without any issue (except animation freeze  when SPIFFS serves big web page, but's it normal, it's sync), working with latest Arduino-ESP8266 repo (SDK 1.5.2)

And @Makuna, I played with animation class and definition, your piece of code is just amazing, thanks a lot for you great work,...

Also for those who want to drive WS2812 ESP8266 3V3 with 5V without level shifter, look at this trick
[image]

It's not my idea, original post [here](
http://www.electrobob.com/ws2812-level-translator/) this is efficient and so simple, just a diode on 1st LED and as I always got one WS2812 led on all my boards, it's not an issue

Makuna, 2016-04-06:
BTW, some of the newer chips (4 element RGBW) are far more tolerant of the 3.3v IO and will work without a level shifter.  

Since you have come back stated it works, I am closing this issue.

closed 2016-04-06 by Makuna

hallard, 2016-04-06:
Oh, I received the RGBW today, I will test ;-)

TotallyInformation, 2016-04-06:
Not had chance to play yet I'm afraid but I will do. I've no doubt you've fixed it though given the comments. Thanks so much for digging into the issue and solving it for us. Your library is definitely cool & I've had the random fade running on an Arduino Nano on my desk for a few days, very relaxing. :-)

@hallard thanks for the testing & the info, very useful.