Makuna/NeoPixelBus #90: NeoPixelAnimator branch of NeoPixelBus seems to be gone

morphygee, 2016-04-02:
Hi,

is it possible to bring it back? There is a project that use it:
https://github.com/Makuna/NeoPixelBus/issues/50

Makuna, 2016-04-02:
Unfortunately no. There was a notice for months now that all the secondary branches where going away and that master would replace them.  

That branch evolved into what Master is today.  The animator is present (and the wiki now documents it).  The master branch supports all common platforms including the Esp8266.

It should be pretty simple to convert to master, the key two things is that the definition of the strip is slightly different and that the animation update method now takes a structure that contains the progress value (and other useful values) instead of just the progress.  That should be about it.

If you have questions about converting, jump on gitter and ask away.
https://gitter.im/Makuna/NeoPixelBus

morphygee, 2016-04-02:
thank you for the info ;)

closed 2016-04-03 by Makuna