Makuna/NeoPixelBus #103: Dynamic change of NeoPixel Type ?

hallard, 2016-04-25:
Following dynamic strip example FAQ here https://github.com/Makuna/NeoPixelBus/wiki/FAQ-%233, is there also a way of changing NeoPixel Type avoiding x pointer declaration ?
for example change from `NeoGrbFeature` to `NeoRgbFeature`

IE something like that ?
```
  // delete the previous dynamically created strip
  if (rgb_led != NULL) 
     delete rgb_led; 

  if (config.led_type == RGB_TYPE_GRB ) {
    rgb_led = new NeoPixelBus<NeoGrbFeature, Neo800KbpsMethod>(config.led_num, config.led_gpio); 
  } else if (config.led_type == RGB_TYPE_RGBW ) {
    rgb_led = new NeoPixelBus<NeoRgbwFeature, Neo800KbpsMethod>(config.led_num, config.led_gpio); 
  } else if (config.led_type == RGB_TYPE_GRBW ) {
    rgb_led = new NeoPixelBus<NeoGrbwFeature, Neo800KbpsMethod>(config.led_num, config.led_gpio); 
  } else {
    rgb_led = new NeoPixelBus<NeoRgbFeature, Neo800KbpsMethod>(config.led_num, config.led_gpio); 
  }
  
  rgb_led->Begin();

```

Thank for any help

Makuna, 2016-04-25:
This is an embedded device, I have got to ask why you are trying to do that?  The hardware generally doesn't change dynamically; and this is basically what you are asking for.

hallard, 2016-04-26:
@Makuna 
I agree on hardware side, but in my case, I'm dealing with lot of testing and different boards with different WS2812 chip from various manufacturers. Some time I don't even know what kind of module it is until I'm testing them (RGB ot GRB) depending on suppliers.
I've got one generic config interface for all my modules and I'm able to do configuration of strip (#of led, type and GPIO pin) as follow 

[image]

I wanted doing this to avoid changing declaration of NeoPixel type on each compilation depending on board I'm using (because this is what I'm doing today, and each time I got the wrong one or I forgot to change it (because I'm working on so much different boards same time)
Anyway your answer is what I suspected, not possible without changing lot of things. And even if my C knowledge if strong, on C++ side, I'm far from a specialist so I will continue with compilation ;-)

Thanks for you explanation

Makuna, 2016-04-26:
You can take this and expand upon the included wrapper class, it doesn't have all the types, but it does work.

```
#include <NeoPixelAnimator.h>
#include <NeoPixelBus.h>

enum NeoPixelType
{
  NeoPixelType_None,
  NeoPixelType_Rgb,
  NeoPixelType_Rgbw
};

class NeoPixelWrapper
{
public:
  NeoPixelWrapper() :
      // initialize each member to null
      _pRgb(NULL),
      _pRgbw(NULL),
      _type(NeoPixelType_None)
  {
    
  }

  ~NeoPixelWrapper()
  {
    cleanup();
  }
  
  void Begin(NeoPixelType type, uint16_t countPixels, uint8_t pin)
  {
    cleanup();
    _type = type;
    
    switch (_type)
    {
      case NeoPixelType_Rgb:
        _pRgb = new NeoPixelBus<NeoRgbFeature,Neo800KbpsMethod>(countPixels, pin);
        _pRgb->Begin();
        break;
      case NeoPixelType_Rgbw:
        
        _pRgbw = new NeoPixelBus<NeoRgbwFeature,Neo800KbpsMethod>(countPixels, pin);NULL;
        _pRgbw->Begin();
        break;
    }
  }

  void Show()
  {
    switch (_type)
    {
      case NeoPixelType_Rgb:
        _pRgb->Show();
        break;
      case NeoPixelType_Rgbw:
        _pRgbw->Show();
        break;
    }
  }
  bool CanShow() const
  {
    switch (_type)
    {
      case NeoPixelType_Rgb:
        _pRgb->CanShow();
        break;
      case NeoPixelType_Rgbw:
        _pRgbw->CanShow();
        break;
    }
  }

  // replicate all the calls like the above
    bool IsDirty() const
    {
    }

    void Dirty()
    {
    }

    void ResetDirty()
    {
    }

    uint8_t* Pixels() const
    {
    }

    size_t PixelsSize() const
    {
    }

    size_t PixelSize() const
    {
    }

    uint16_t PixelCount() const
    {
    }

    void SetPixelColor(uint16_t indexPixel, RgbColor color)
    {
      switch (_type)
      {
        case NeoPixelType_Rgb:
          _pRgb->SetPixelColor(indexPixel, color);
          break;
        case NeoPixelType_Rgbw:
          _pRgbw->SetPixelColor(indexPixel, color);
          break;
      }
    }

    void SetPixelColor(uint16_t indexPixel, RgbwColor color)
    {
      switch (_type)
      {
        case NeoPixelType_Rgb:
          // doesn't support it so we don't call it
          break;
        case NeoPixelType_Rgbw:
          _pRgbw->SetPixelColor(indexPixel, color);
          break;
      }
    }

    RgbColor GetPixelColor(uint16_t indexPixel) const
    {
      switch (_type)
      {
        case NeoPixelType_Rgb:
          return _pRgb->GetPixelColor(indexPixel);
          break;
        case NeoPixelType_Rgbw:
          // doesn't support it so we don't return it
          // ends up returning black below
          break;
      }
      return 0;
    }

// NOTE:  Due to feature differences, some support RGBW but the method name
// here needs to be unique, thus GetPixeColorRgbw
    RgbwColor GetPixelColorRgbw(uint16_t indexPixel) const
    {
      switch (_type)
      {
        case NeoPixelType_Rgb:
          return _pRgb->GetPixelColor(indexPixel);
          break;
        case NeoPixelType_Rgbw:
          return _pRgbw->GetPixelColor(indexPixel);
          break;
      }
      return 0;
    }

    void ClearTo(RgbColor color)
    {
    }

    void ClearTo(RgbwColor color)
    {
    }
    
    void ClearTo(RgbColor color, uint16_t first, uint16_t last)
    {
    }

    void ClearTo(RgbwColor color, uint16_t first, uint16_t last)
    {
    }

    void RotateLeft(uint16_t rotationCount)
    {
    }

    void RotateLeft(uint16_t rotationCount, uint16_t first, uint16_t last)
    {
    }

    void ShiftLeft(uint16_t shiftCount)
    {
    }

    void ShiftLeft(uint16_t shiftCount, uint16_t first, uint16_t last)
    {
    }

    void RotateRight(uint16_t rotationCount)
    {
    }

    void RotateRight(uint16_t rotationCount, uint16_t first, uint16_t last)
    {
    }

    void ShiftRight(uint16_t shiftCount)
    {
    }

    void ShiftRight(uint16_t shiftCount, uint16_t first, uint16_t last)
    {
    }
    
private:
  NeoPixelType _type;

  // have a member for every possible type
  NeoPixelBus<NeoRgbFeature,Neo800KbpsMethod>* _pRgb;
  NeoPixelBus<NeoRgbwFeature,Neo800KbpsMethod>* _pRgbw;

  void cleanup()
  {
    switch (_type)
    {
      case NeoPixelType_Rgb:
        delete _pRgb;
        _pRgb = NULL;
        break;
      case NeoPixelType_Rgbw:
        delete _pRgbw;
        _pRgbw = NULL;
        break;
    }
  }
};

NeoPixelWrapper strip;

void setup() {
  // start with Rgb with 10 pixels
  strip.Begin(NeoPixelType_Rgb, 10, 2);
  strip.Show();

  // switch to rgbw with 12 pixels
  strip.Begin(NeoPixelType_Rgbw, 12, 2);
  strip.Show();
}

void loop() {
  // put your main code here, to run repeatedly:

}
```

hallard, 2016-04-26:
@Makuna 
This is great I will try this and may be I'll have more question later, I also need to integrate @me-do-dev async TCP  ;-)

hallard, 2016-04-27:
@Makuna 
Got it, working like a charm

FYI here the `NeoPixelWrapper.h`, I added the pixel color for HSL. I'm not really sure on what need to be commented into `GetPixelColor()` and `SetPixelColor()` depending on NeoPixel type but seems to works

My next problem is the dynamic NeoPixelAnimator object and MyAnimationState array, because it can change depending on # of RGB led but I don't know how to wrap it.
But I declared AnimCount the maximum size I need, not clean but works, don't worry about this one

```arduino
#ifdef RGB_LED
  NeoPixelWrapper rgb_led; // The wrapped strip
  const uint16_t AnimCount = 17 * 2 / 5 + 1; // animations for the tail and one extra
  uint8_t  rgb_led_front_pixel;   // general purpose variable used to store pixel index
  uint8_t  rgb_led_effect_state;  // general purpose variable used to store effect state
  RgbColor rgb_led_color; // RGB Led color
  NeoGamma<NeoGammaTableMethod> colorGamma; // for any fade animations

  // since we can have RGB or RGBW strip we need to have both declaration
  struct MyAnimationState   {
    RgbwColor RgbwStartingColor;
    RgbwColor RgbwEndingColor;
    RgbColor  RgbStartingColor;
    RgbColor  RgbEndingColor;
    uint8_t IndexPixel;   // general purpose variable used to store pixel index
  };
  NeoPixelAnimator animations(AnimCount); // NeoPixel animation management object
  MyAnimationState animationState[AnimCount];
#endif
```

Thank you very much for your help. and here the new  `NeoPixelWrapper.h` I'm using 

```arduino
#include <NeoPixelAnimator.h>
#include <NeoPixelBus.h>

enum NeoPixelType
{
  NeoPixelType_None = 0,
  NeoPixelType_Rgb  = 1,
  NeoPixelType_Grb  = 2,
  NeoPixelType_Rgbw = 3,
  NeoPixelType_Grbw = 4,
  NeoPixelType_End  = 5
};

#define Is_RGBW(l) ( l==NeoPixelType_Rgbw || l==NeoPixelType_Grbw )

class NeoPixelWrapper
{
public:
  NeoPixelWrapper() :
      // initialize each member to null
      _pRgb(NULL),
      _pGrb(NULL),
      _pRgbw(NULL),
      _pGrbw(NULL),
      _type(NeoPixelType_None)
  {

  }

  ~NeoPixelWrapper()
  {
    cleanup();
  }

  void Begin(NeoPixelType type, uint16_t countPixels, uint8_t pin)
  {
    cleanup();
[... 171 lines cut ...]

    void ShiftRight(uint16_t shiftCount, uint16_t first, uint16_t last)
    {
    }

private:
  NeoPixelType _type;

  // have a member for every possible type
  NeoPixelBus<NeoRgbFeature,NeoEsp8266BitBang800KbpsMethod>*  _pRgb;
  NeoPixelBus<NeoGrbFeature,NeoEsp8266BitBang800KbpsMethod>*  _pGrb;
  NeoPixelBus<NeoRgbwFeature,NeoEsp8266BitBang800KbpsMethod>* _pRgbw;
  NeoPixelBus<NeoGrbwFeature,NeoEsp8266BitBang800KbpsMethod>* _pGrbw;

  void cleanup()
  {
    switch (_type) {
      case NeoPixelType_Rgb:  delete _pRgb ; _pRgb  = NULL; break;
      case NeoPixelType_Grb:  delete _pGrb ; _pGrb  = NULL; break;
      case NeoPixelType_Rgbw: delete _pRgbw; _pRgbw = NULL; break;
      case NeoPixelType_Grbw: delete _pGrbw; _pGrbw = NULL; break;
    }
  }
};
```

Makuna, 2016-04-27:
You should not need the HslColor methods.   Behind the scenes, all the color objects will autoconvert to RgbColor; this is how the library works.  The reason Rgbw is present is that while converting from 3 color element like Hsl to a four color element like Rgbw is automatic; converting back is a loss of information so it is not automatic.

hallard, 2016-04-27:
Humm, without HSL I was getting a compiler error, I'll try back tomorrow, I need to go, thanks for your help.

closed 2016-05-22 by Makuna

zigi-dz, 2021-02-22:
hey can someone help me to modifie it for led5050

Makuna, 2021-02-23:
@zigi-dz Commenting on a closed issue is not a valid way to get help.  Check out the ReadMe or the Wiki for pointers to areas you can ask questions.  Github Discussions, Gitter, and Discord.