Makuna/NeoPixelBus #105: How to create slow animations?

msurguy, 2016-05-30:
Hello and thanks for making this library!

I have been trying to create animations that last 2-5 minutes and so far haven't been able to do that with this library. For example trying to set the time to 60000 at https://github.com/Makuna/NeoPixelBus/blob/master/examples/NeoPixelFunFadeInOut/NeoPixelFunFadeInOut.ino#L87 results in very jittery animation. 

Is there a suggested way to create animations that last more than a minute? 

Thanks!

Makuna, 2016-05-30:
Referring to the [NeoPixelAnimator API](https://github.com/Makuna/NeoPixelBus/wiki/NeoPixelAnimator-object-API), the default time scale is milliseconds, with a time value of uint16, this means the max is about 65.535 seconds.  You will need to use NEO_CENTISECONDS  to extend this to just under 11 minutes but still give you 100 "possible" updates per seconds.

What do you mean by jittery?  The colors are "stair stepping" where you can see the color change every second or two? Or are they actually jumping around a lot?

Note, For quick questions you jump on Gitter and ask away.  
[image]

msurguy, 2016-05-31:
I'll move this convo over to Gitter, thanks!

closed 2016-05-31 by msurguy