Makuna/NeoPixelBus #106: using NeoPixelBus for APA102 LED strip

mkeyno, 2016-06-01:
hi @Makuna , is it possible to use you lib on ESP module  for the LED with clock and data such as APA102? I've tried to use FastLED lib but no success with APA102, whereas had good result on APA104

Makuna, 2016-06-01:
The APA104 is an improved WS2812B clone, still a single data wire bus.
The APA102 is has CLK and DI lines, so a two wire bus.
They are not "bus" compatible.

It could be done, but it is truly no longer a "NeoPixel Bus" which is what this library is all about, its a "Dot Star Bus" (in the Adafruit nomenclature).

I don't have or use any of these class of LEDs.

mkeyno, 2016-06-01:
but it seems run  APA102  is easier than bitbang APA104 , why don't you port it , I do lot of fun with your lib and APA104, it was surprise me your lib not support  two wire bus.

Makuna, 2016-06-07:
I am leaving this open to track this.  At the moment I don't have any to even test against.

Makuna, 2016-08-30:
I am actively working on this support now.  If you would interested in testing it let me know.

mkeyno, 2016-09-01:
very glad to hear your decision  Michel , my LED project with ESP held back for lack of this  feature , your lib has very powerful logic  in understand  animation but suffer form support high speed LED chip , I've tried to use following code but the result was unsatisfied, as you well know the HW SPI in ESP can reach to 80Mhz and easily can run 24Mhz for fast LED chip like APA102, I eagerly follow up your lib and commit as before dear Michel  

```cpp
void inline APA102_fast_SPI(  RGB *_leds, uint16_t led_num=NUM_LEDS)
{
  uint16_t i;
  uint8_t *rawarray=(uint8_t*)_leds;
  //SPI_init();

  SPI.transfer(0x00);  // Start Frame
  SPI.transfer(0x00);
  SPI.transfer(0x00);
  SPI.transfer(0x00);

  for (i=0; i<(3*led_num); i+=3)
  {
    SPI.transfer(0xff);  // Maximum global brightness
    SPI.transfer(rawarray[i+0]);
    SPI.transfer(rawarray[i+1]);
    SPI.transfer(rawarray[i+2]);
  }

  // End frame: 8+8*(leds >> 4) clock cycles    
  //for (i=0; i<leds; i+=16)
for (int i = 0; i < (1 + led_num / 32) * 4; i++)      
  {
    SPI.transfer(0xff);  // 8 more clock cycles
  }
}
```

Makuna, 2016-09-01:
What color order are your LEDs?  

The ones I have are BGR.

closed 2016-09-02 by Makuna

mkeyno, 2016-09-02:
@Makuna  I set the order as ```#define COLOR_ORDER GRB``` and set ```SPI.setFrequency(20000000); ///< 20MHz``` , I've also tried other speed , but it seems colors is not connected together as it was when using APA104 with your lib