Makuna/NeoPixelBus #109: Enhancement: grant its own library to color handling classes

unaiur, 2016-06-07:
You have write an excelent color handling class collection, like no other in Arduino world. They deserve its own library so other projects can reuse it.

Also, NeoPixelAnimator doesn't seem to depend on NeoPixelBus and can be used with other addressable led strips, so maybe could also get its own library too.

Makuna, 2016-06-07:
Great points and these are things I have considered; but until Arduino supports some sort of automatic dependency library support I can't move the color objects out due to the core work relying on it.

I don't want to burden the less technical users with multiple libraries to just get it to work.

I will leave this open incase the Arduino IDE ever supports automatic library dependency support.

jamesmyatt, 2020-01-27:
You could solve the dependency issue with PlatformIO

ben-mkiv, 2020-02-11:
couldn't this be solved by having a separate github project for them which the neopixelbus project depends on, so you could build both on your own (while the neopixel lib still includes the color library)

`#ifdef` annotations in the headers should avoid loading it twice in projects

Makuna, 2020-02-11:
The required feature (dependant libraries) was just added to Arduino to support it.  So it would require the user to be using the latest build (at this time, 1.8.10) to get the support.

"
*When I install a library that I know depends on another library, will this other library be installed as well?*
You can specify the dependencies of your library in the depends field of library.properties. As of Arduino IDE 1.8.10, the user will be prompted to install those libraries when they install your library via Library Manager. For more information, see [the library.properties file format documentation](https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5:-Library-specification#libraryproperties-file-format).
"