Makuna/NeoPixelBus #111: NeoGrbFeature and Neo800KbpsMethod

promd, 2016-06-11:
Hi, I can't make it work.....
Always getting an Error 
"Type 'NeoGrbFeature' could not be resolved" 
and 
"Type 'Neo800KbpsMethod' could not be resolved".

on line
`NeoPixelBus<NeoGrbFeature,Neo800KbpsMethod> strip(pixelCount, pixelPin);`

And yes, I can't see them defined anywhere.
It seems to be something is missing from the sources ?? 
I am on platform.io, for platform espressif (esp12).

Makuna, 2016-06-11:
Your code or a sample?  Please include the sketch or a reference to the example.
I do not use nor test with platform io.  I tried to set it up once and got too many failures during installations.  This library was written for Arduino.

Is ARDUINO_ARCH_ESP8266 defined in your build?

promd, 2016-06-11:
Nevermind. Seems they are using a rather old version of your Library. I did clone from Git, and I am running now - Thx!

closed 2016-06-11 by promd