Makuna/NeoPixelBus #113: very slow frame rate for ESP & APA102

mkeyno, 2016-06-27:
hi @Makuna , I'm so sorry to draw issue for my question but I really appreciated for your help and opinion , actually I'm trying to use fastLED lib to drive my APA102 for POV project , I've tried to implement it with ESP, but frame rate is so slow for 144 APA102 pixel,
my approach is to read pixel data from SPIFF file without closing it , but result beyond my expectation very slow even with 160Mhz clock
following is the serial output and loop code
i=0 New Image=1.bin duration=4
[frame]1076598
[frame]1043446
[frame]1044003
[frame]1043555
i=1 New Image=2.bin duration=4
[frame]1057822
[frame]1052909
[frame]1052453
i=2 New Image=3.bin duration=4
[frame]1067164
[frame]1042669
[frame]1043549
[frame]1042468
i=3 New Image=4.bin duration=4
[frame]1059210
[frame]1044297
[frame]1044701
[frame]1044550
i=4 New Image=5.bin duration=4
[frame]1059668
[frame]1042478
[frame]1043678
[frame]1043824
i=5 New Image= duration=0
FILE NOT OPEN
i=0 New Image=1.bin duration=4
[frame]1075868
[frame]1044087
[frame]1044103
[frame]1044578
i=1 New Image=2.bin duration=4
[frame]1059307
[frame]1052474
[frame]1052181
i=2 New Image=3.bin duration=4
[frame]1067527
[frame]1042757
[frame]1042843
[frame]1043076
i=3 New Image=4.bin duration=4
[frame]1058051
[frame]1045418
[frame]1046048
[frame]1045034
i=4 New Image=5.bin duration=4
[frame]1059883
[frame]1043249
[frame]1043334
[frame]1043710
i=5 New Image= duration=0
FILE NOT OPEN
i=0 New Image=1.bin duration=4

loop code
void loop()
{

if(!SHOW) server.handleClient();

webSocket.loop();

if(SHOW)
{
if( (millis()- OpenlastTime) > DURATION[image_index]*1000L )
{
FILe.close();
image_index++;
if(image_index>IMAGE_NUM) image_index=0;

       Serial.print(F(" i="));  Serial.print(image_index); 
      Serial.print(F(" New Image="));  Serial.print(IMAGES[image_index]);
      Serial.print(F(" duration="));  Serial.println(DURATION[image_index]);

       FILe=SPIFFS.open("/SHOW/"+IMAGES[image_index], "r");
      if(!FILe) Serial.println("FILE NOT OPEN");
      OpenlastTime=millis();
    }
    if(!Start_new_round && (micros()-lastLineShow)> lineInterval)
                      {
                      int i=0,j=0;
                      byte rgb=0,r,g,b; 
                      while(FILe.available()&& i< (NUM_LEDS*3))
                                                                {
                                                                byte c=FILe.read();
                                                                     if(rgb==0) r=byte(c);
                                                                else if(rgb==1) g=byte(c);
                                                                else if(rgb==2) b=byte(c);
                                                                else           { rgb=0;
                                                                                 leds[j] = CRGB(r,g,b);
                                                                                 j++; 
                                                                                 }
                                                                rgb++;            
                                                                i++;
                                                                }
                      FastLED.show(); lastLineShow=micros();                          
                      Current_imageLine++;                                        
                      }
   FastLED.show(); lastLineShow=micros(); }
      if(Current_imageLine >= IMAGES_LINES)   
      { 
        Serial.print("[frame]"); Serial.println((micros()-lastFrameShow));
        Current_imageLine = 0;
        lastFrameShow=micros(); 
        Start_new_round=false; 
        FILe.close();
        FILe=SPIFFS.open("/SHOW/"+IMAGES[image_index], "r");
       } 

  }

}

Makuna, 2016-06-28:
This has nothing to do with this NeoPixelBus library and is an inappropriate issue to enter here.

I would suggest either the [esp8266 Arduino Gitter](https://gitter.im/esp8266/Arduino) or something similar for FastLed library that you are having an issue with.

closed 2016-06-28 by Makuna

mkeyno, 2016-06-28:
thanks @Makuna  yes I know this is nothing to do with your repo  but I was hoping to  have your idea to what is the best method to read image file lines an fed it to LED and then show it , if you interested I could explain more detail about my approach
thanks anyway

Makuna, 2016-06-28:
@mkeyno Again, look at gitter link, its a chat system tied to github repos, a better approach to asking this sort of thing and has a wider audience of people of know details of SPIFFs that help you.

mkeyno, 2016-06-29:
I'm sorry to be unintentional rube Michel and I was aware for this feature in your lib but as you said before yet not support 4 wire LED chip and that's why I thinking maybe you can share how to handle fast file reading for such purpose , any way  I apologize again