Makuna/NeoPixelBus #118: Add sRGB support and white calibration table.

far5893, 2016-07-21:
Please add support for sRGB color space  (classical table with x^2.2 pwm values)
and white color calibration table  to compensate different color intensities on chep cinese rgb strip.

Makuna, 2016-07-21:
Please provide more details.

What color problem are you seeing? Do you have pointers to discussions on the issue?  Or are you just talking about gamma correction that is already supported?

far5893, 2016-07-22:
OK what you call gamma correction is like sRGB support , i see your formula.

About white balance i didn't found anything  , my strip are more blue   (but i think is normal on cheap 
strip because red led is little shifted on  orange).
You can call white balance or Color temperature, if you like, and is a fixed correction in ratio between the three channels.

Makuna, 2016-07-22:
This is going to be specific to  run of leds, so without the leds and without a color balance sensor I can't build a table/formula to correct it.  
It would work similar to the gamma, but a seperate entry for channel.

far5893, 2016-07-25:
White balance is a simple multiply constant applied to three channels to change relative weights,
it's the user that needed to change or set it and it's the user that need the color balance sensor (or a very good eye)!

Makuna, 2016-07-25:
You are assuming the relative balance is the same linearly across all brightness's of each sub LED element (R, G, B); but you can't assume this.

far5893, 2016-07-25:
PWM is on-off tecnology and intrinsecally linear if switch on-off(in nanoseconds range for LED's) delay is neglegible , so Yes it's linear.

Makuna, 2019-03-31:
I have yet to come up with a solution that makes sense for me to implement versus the user to just do the work.  A solution has to be no less understandable and no less complex in its use that a purely user supplied code.

Here is an example, this handles a static ratio across all levels, where the correction value of 255 in a channel means no change, and lower values means to lower the brightness.

```
class NeoColorBalance
{
public:
    NeoColorBalance(RgbColor correction) :
        _correction(correction)
    {
    }

    RgbColor Correct(const RgbColor& original)
    {
        return RgbColor(original.R * _correction.R / 255,
            original.G * _correction.G / 255,
            original.B * _correction.B / 255);
    }

private:
    RgbColor _correction;
};
```