Makuna/NeoPixelBus #119: Using NeoEsp8266AsyncUart800KbpsMethod> and changing calling Serial.begin() again causes wdt reset.

sticilface, 2016-07-22:
This behaviour is new since commit https://github.com/esp8266/Arduino/pull/2239, which causes wdt reset whenever serial.begin() was called again, however, this has been fixed in https://github.com/esp8266/Arduino/pull/2307.   Prior to all these commits NeoEsp8266AsyncUart800KbpsMethod worked fine.  Currently DMA and normal UART work fine, but using async and calling serial.begin() again, even with the same serial speed, and making the stip buffer dirty, causes a wdt.  @me-no-dev has suggested that the RX buffer interrupt that is now being used is shared between both Serial and Serial1.   

sketch below causes it

```cpp
#include <NeoPixelBus.h>

//NeoPixelBus<NeoGrbFeature, Neo800KbpsMethod> strip(20, 2); // does not wdt 
//NeoPixelBus<NeoGrbFeature, NeoEsp8266Uart800KbpsMethod> strip(20, 2); //  does not wdt

NeoPixelBus<NeoGrbFeature, NeoEsp8266AsyncUart800KbpsMethod> strip(20, 2); //  causes wdt

bool triggered = false; 
uint32_t timer = 0; 
 
void setup() {
  
  Serial.begin(115200);
  Serial.println("BEGIN 115200"); 
  strip.Begin();
  strip.Show();
    
}

void loop() {

  if (millis() - timer > 500) {
    Serial.println("working.....");
    timer = millis(); 
  }

  if (!triggered && millis() > 5000) {
    
  if (Serial) {
    Serial.println("Changing Serial\n\n");
    Serial.flush();
    Serial.begin(115200);
  }

  strip.Dirty(); // this is important.. otherwise it works.  can set a color too... 
  strip.Show(); 
  triggered = true;
  }
        
}

```

Makuna, 2016-09-03:
In general, calling Serial.begin() will reset hardware state which Dma and Uart methods rely on.  For DMA method it is documented that you can't call Serial.begin() after calling NeoPixelBus.Begin() due to this.

The work around is to destruct the NeoPixelBus before calling Serial.begin(), and then recreate it afterwards.

Makuna, 2019-01-07:
The fix for this requires changes in the Esp8266 hardware serial core code to expose a model of sharing the single shared interrupt.  This work is actively under construction.

Makuna, 2019-01-07:
New branch created with changes to match pull request to esp8266/Arduino 
https://github.com/Makuna/NeoPixelBus/tree/SharedUartIsr