Makuna/NeoPixelBus #121: Wiki confusion NeoEsp8266Dma800KbpsMethod vs ESP-01

penfold42, 2016-07-25:
from the NeoEsp8266Dma800KbpsMethod section:

"It also requires the use of the RDX0/GPIO3 pin. This pin is not exposed on the ESP-01. The normal feature of this pin is the "Serial" receive."

Are you sure its not exposed on the ESP-01 ?

Obviously this will cause issues if you have eg an ftdi programmer attached cos they will both try to drive the RXD line so the general warnings still apply

Makuna, 2016-07-25:
ESP-01, the serial receive is actually the RDX1, but the Serial has this `Swap()` method to allow using it and I believe on Esp-01 it does this default.  RDX0 and RDX1 are low level hardware references, most esp boards leave off number and assume you are only using one; which causes the confusion.

penfold42, 2016-07-26:
Thanks - for the library and your quick reply.

are you sure you're really really sure? *grin*
I just tested it on an ESP-01 and it appears to work fine.

Using library NeoPixelBus_by_Makuna at version 2.1.4

Here's my setup line:
NeoPixelBus<NeoGrbFeature, NeoEsp8266Dma800KbpsMethod> strip(PixelCount, NeoPin);

My 6 led test strip is now attached to the RXD line of the ESP-01.
My homebrew FTDI programmer had to be modified to put a resistor between FTDI.TX and ESP.RX

thanks!

Makuna, 2016-07-26:
It didn't work 9 months ago!  

Ok, now I have to go back and test it again since you stated it does work.

Can you link me the exact board you have?

penfold42, 2016-07-26:
One of the $2 blues ones from eBay like this one:
https://www.ebay.com.au/itm/351762665665

I upgraded the flash chip on it but I can't see that making a difference.

I started with a 4k7 resistor between the ftdi and RXD but it was too high and uploads didn't work.
I then added a 1k2 resistor in parallel and then uploads worked

penfold42, 2016-07-27:
Btw I also  tested with 316 LEDs attached driven by Hyperion sending udp packets.

For the most part it's fantastic. There's the odd glitch but that could just be the lack of a level converter.

There's also a laggy smithing effect going on but that could be a Hyperion issue, the fact the sketch is setup for 750 LEDs or just simply a high UDP rate

penfold42, 2016-07-27:
Glitches I think are fixed - 74hct138 as a level shifter send to have cured it.

Laggy fixed too - Hyperion bug where smoothing "none" isn't honored

Makuna, 2016-12-19:
Fixed the WIKI.  THANKS!

closed 2016-12-19 by Makuna