Makuna/NeoPixelBus #122: Esp8266 async uart direct pixel access problems.

penfold42, 2016-08-02:
My sketch listens for UDP led data and simply sets the string elements directly.

With the non async uart it works fine.

With the async uart it seems to ignore every second update.

Has this been seen before ?

I tried reproducing it in a much simpler sketch but the strip wouldn't update at all

Makuna, 2016-08-02:
Can you provide the simple sketch that doesn't update at all?

How often is your udp sketch trying to call 'show ()'?
Where does your sketch call show, a network call back, within loop()?

penfold42, 2016-08-02:
Thanks for the quick reply.

I will, but I want to make it closer to my main project and not the quick hack I did to rule of stupidity on my part !

The main project is running I loop it roughly looks like this:
Loop()
Check for packet ()
Parsepacket()
UDP.read() into my buffer
For() to set the pixels
Dirty()
Show()

So it'll only call show once per packet.
I was sending packets with Hyperion and then very slowly - once a second. Both show the issue.

Makuna, 2016-09-02:
Is the Show() being called within the context of a network callback?

penfold42, 2016-09-02:
Nope - I just poll for packets.

If you're interested, the code is here:
https://github.com/penfold42/stuff/blob/master/ESP-udp-neo-OTA-webconfig.ino

Makuna, 2016-12-21:
Did you confirm that the following code is packed correctly (is really only 3 bytes) as the compiler can make them 4 bytes at will unless marked with a pragma packed.
```
struct GrbPixel {

    uint8_t G;

    uint8_t R;

    uint8_t B;

};

GrbPixel* pixels = (GrbPixel*) strip.Pixels();
```

penfold42, 2016-12-21:
I'll dig out the code and check.

Wouldn't that fail with the non async mode too if that were the cause ?

Makuna, 2016-12-26:
So I tried this minimal sketch that tests alternating frames.  I am not seeing missed frames.

```
#include <NeoPixelBus.h>

// use a larger pixel count to cause it to
// make the update take longer and exercise
// the async features of DMA and AsyncUart
const uint16_t PixelCount = 64; // 128 // 256 // 512

#define colorSaturation 128

// three element pixels, in different order and speeds
NeoPixelBus<NeoGrbFeature, NeoEsp8266AsyncUart800KbpsMethod> strip(PixelCount);
//NeoEsp8266Uart800KbpsMethod
//Neo800KbpsMethod
// NeoEsp8266AsyncUart800KbpsMethod

RgbColor red(colorSaturation, 0, 0);
RgbColor green(0, colorSaturation, 0);
RgbColor blue(0, 0, colorSaturation);
RgbColor white(colorSaturation);
RgbColor black(0);

uint8_t frame;

void setup()
{
    Serial.begin(115200);
    while (!Serial); // wait for serial attach

    Serial.println();
    Serial.println("Initializing...");
    Serial.flush();

    // this resets all the neopixels to an off state
    strip.Begin();
    strip.Show();

    Serial.println();
    Serial.println("Running...");

    frame = 0;
}

void loop()
{
    // without the delay, this will run as fast as it can
    // 
    //delay(1000);

    // set the colors, 
    // use alternate frames to red and blue
    // if a frame is skipped consistently, one color wont show
    if (frame % 2 == 0) // even frames draw red
      strip.SetPixelColor(0, red);
    if (frame % 2 == 1) // odd frames draw blue
      strip.SetPixelColor(1, blue);
    strip.Show();

    // set both back to black for the next frame
    // faster than clearing
    strip.SetPixelColor(0, black);
    strip.SetPixelColor(1, black);

    frame++;
}

```

Makuna, 2016-12-28:
I think I found the problem you are running into.

You are caching the Pixels() pointer globally.  For the AsyncUart the value returned from Pixels alternates between two buffers when the Show is called.  This is not documented but is present in a comment in the code.

For the AsyncUart you are required to request the Pixels() again after a Show.  

Generally it is not good to store anything globally that you can easily retreive.

Makuna, 2016-12-28:
Updated the Wiki to make this issue more discoverable.  See [NeoEsp8266AsyncUart800KbpsMethod](https://github.com/Makuna/NeoPixelBus/wiki/NeoPixelBus-object#neoesp8266asyncuart800kbpsmethod)

closed 2016-12-28 by Makuna

penfold42, 2016-12-28:
Well spotted - thanks!

Sorry i didn't have a chance to re-investigate