Makuna/NeoPixelBus #134: NeoBitmapFile SPIFFS support

sfranzyshen, 2016-09-27:
I attempted to port this to SPIFFS but failed ... any insight?

Makuna, 2016-09-27:
(using gitter is a better approach for this issue as its more of a question).  See readme or wiki for link to gitter.

Please provide more details.  What do you mean "port"?  Sample declaration will help.  A description of the failure (compile?  runtime?) with important details is useful.

Makuna, 2016-09-27:
see https://github.com/esp8266/Arduino/issues/2553

sfranzyshen, 2016-09-27:
I tried to port the examples/NeoPixelBitmap/NeoPixelBitmap.ino to use FS.h instead of SD.h and I made edits to use SPIFFS instead of SD

```
@@ -11,19 +11,19 @@
 
 #include <NeoPixelBus.h>
 #include <NeoPixelAnimator.h>
-#include <SPI.h> 
-#include <SD.h> 
+//#include <SPI.h> 
+//#include <SD.h> 
+#include <FS.h>
+//const int chipSelect = 13; // make sure to set this to your SD carder reader CS
 
-const int chipSelect = D8; // make sure to set this to your SD carder reader CS
-
-//typedef NeoGrbFeature MyPixelColorFeature;
-typedef NeoGrbwFeature MyPixelColorFeature;
+typedef NeoGrbFeature MyPixelColorFeature;
+//typedef NeoGrbwFeature MyPixelColorFeature;
 
 const uint16_t PixelCount = 144; // the sample images are meant for 144 pixels
 const uint16_t PixelPin = 2;
 const uint16_t AnimCount = 1; // we only need one
 
-NeoPixelBus<MyPixelColorFeature, Neo800KbpsMethod> strip(PixelCount, PixelPin);
+NeoPixelBus<MyPixelColorFeature, NeoEsp8266Uart800KbpsMethod> strip(PixelCount, PixelPin);
 // for esp8266 omit the pin
 //NeoPixelBus<MyPixelColorFeature, Neo800KbpsMethod> strip(PixelCount);
 NeoPixelAnimator animations(AnimCount); // NeoPixel animation management object
@@ -59,7 +59,7 @@
     Serial.print("Initializing SD card...");
 
     // see if the card is present and can be initialized:
-    if (!SD.begin(chipSelect))
+    if (!SPIFFS.begin())
     {
         Serial.println("Card failed, or not present");
         // don't do anything more:
@@ -68,7 +68,7 @@
     Serial.println("card initialized.");
 
     // open the file
-    File bitmapFile = SD.open("strings.bmp"); 
+    File bitmapFile = SPIFFS.open("strings.bmp", "r"); 
     if (!bitmapFile)
     {
         Serial.println("File open fail, or not present");
```
but I get failure(s) from the NeoBitmapFile.h include file. 
```
In file included from /home/user/Arduino/libraries/NeoPixelBus_by_Makuna/src/NeoPixelBus.h:50:0,
                 from /home/user/Arduino/NeoPixelBitmap/NeoPixelBitmap.ino:12:
/home/user/Arduino/libraries/NeoPixelBus_by_Makuna/src/internal/NeoBitmapFile.h: In instantiation of 'bool NeoBitmapFile<T_COLOR_FEATURE, T_FILE_METHOD>::Begin(T_FILE_METHOD) [with T_COLOR_FEATURE = NeoGrbFeature; T_FILE_METHOD = fs::File]':
/home/user/Arduino/NeoPixelBitmap/NeoPixelBitmap.ino:80:32:   required from here
/home/user/Arduino/libraries/NeoPixelBus_by_Makuna/src/internal/NeoBitmapFile.h:96:22: error: no matching function for call to 'fs::File::seek(int)'
         if (!file || !file.seek(0))
```
Why doesn't the FS & SD act the same way for "File" type ... what am I missing? The wiki states that I should be able to use SPIFFS

`For Esp8266, you may pass a SPIFFS File type allowing this to read from SPIFFs library.`
and
`Valid classes for FILE are any class that exposes the same API as the File class. It does not require that it be the SDFile`

I am using arduino ide 1.6.8 with esp8266 board support 2.3.0 (installed with board manager) and NeoPixelBus  2.2.3 (installed with library manager)

sfranzyshen, 2016-09-27:
besides the nonstandard seek call ... it looks like a problem with read() also ...

```
/home/user/.arduino15/packages/esp8266/hardware/esp8266/2.3.0/cores/esp8266/FS.h:66:10: note:   candidate expects 2 arguments, 1 provided
In file included from /home/user/Arduino/libraries/NeoPixelBus_by_Makuna/src/NeoPixelBus.h:50:0,
                 from /home/user/Arduino/NeoPixelBitmap/NeoPixelBitmap.ino:12:
/home/user/Arduino/libraries/NeoPixelBus_by_Makuna/src/internal/NeoBitmapFile.h:107:16: error: no matching function for call to 'fs::File::read(BitmapFileHeader*, unsigned int)'
         result = _file.read(&bmpHeader, sizeof(bmpHeader));
                ^
/home/user/Arduino/libraries/NeoPixelBus_by_Makuna/src/internal/NeoBitmapFile.h:107:16: note: candidates are:
In file included from /home/user/Arduino/NeoPixelBitmap/NeoPixelBitmap.ino:16:0:
/home/user/.arduino15/packages/esp8266/hardware/esp8266/2.3.0/cores/esp8266/FS.h:59:9: note: virtual int fs::File::read()
     int read() override;
         ^
/home/user/.arduino15/packages/esp8266/hardware/esp8266/2.3.0/cores/esp8266/FS.h:59:9: note:   candidate expects 0 arguments, 2 provided
/home/user/.arduino15/packages/esp8266/hardware/esp8266/2.3.0/cores/esp8266/FS.h:65:12: note: size_t fs::File::read(uint8_t*, size_t)
     size_t read(uint8_t* buf, size_t size);
            ^
/home/user/.arduino15/packages/esp8266/hardware/esp8266/2.3.0/cores/esp8266/FS.h:65:12: note:   no known conversion for argument 1 from 'BitmapFileHeader*' to 'uint8_t* {aka unsigned char*}'
In file included from /home/user/Arduino/libraries/NeoPixelBus_by_Makuna/src/NeoPixelBus.h:50:0,
                 from /home/user/Arduino/NeoPixelBitmap/NeoPixelBitmap.ino:12:
/home/user/Arduino/libraries/NeoPixelBus_by_Makuna/src/internal/NeoBitmapFile.h:116:16: error: no matching function for call to 'fs::File::read(BitmapInfoHeader*, unsigned int)'
         result = _file.read(&bmpInfoHeader, sizeof(bmpInfoHeader));
                ^
/home/user/Arduino/libraries/NeoPixelBus_by_Makuna/src/internal/NeoBitmapFile.h:116:16: note: candidates are:
In file included from /home/user/Arduino/NeoPixelBitmap/NeoPixelBitmap.ino:16:0:
/home/user/.arduino15/packages/esp8266/hardware/esp8266/2.3.0/cores/esp8266/FS.h:59:9: note: virtual int fs::File::read()
     int read() override;
         ^
/home/user/.arduino15/packages/esp8266/hardware/esp8266/2.3.0/cores/esp8266/FS.h:59:9: note:   candidate expects 0 arguments, 2 provided
/home/user/.arduino15/packages/esp8266/hardware/esp8266/2.3.0/cores/esp8266/FS.h:65:12: note: size_t fs::File::read(uint8_t*, size_t)
     size_t read(uint8_t* buf, size_t size);
            ^
/home/user/.arduino15/packages/esp8266/hardware/esp8266/2.3.0/cores/esp8266/FS.h:65:12: note:   no known conversion for argument 1 from 'BitmapInfoHeader*' to 'uint8_t* {aka unsigned char*}'
In file included from /home/user/Arduino/libraries/NeoPixelBus_by_Makuna/src/NeoPixelBus.h:50:0,
                 from /home/user/Arduino/NeoPixelBitmap/NeoPixelBitmap.ino:12:
/home/user/Arduino/libraries/NeoPixelBus_by_Makuna/src/internal/NeoBitmapFile.h: In instantiation of 'bool NeoBitmapFile<T_COLOR_FEATURE, T_FILE_METHOD>::seek(int16_t, int16_t) [with T_COLOR_FEATURE = NeoGrbFeature; T_FILE_METHOD = fs::File; int16_t = short int]':
/home/user/Arduino/libraries/NeoPixelBus_by_Makuna/src/internal/NeoBitmapFile.h:204:28:   required from 'void NeoBitmapFile<T_COLOR_FEATURE, T_FILE_METHOD>::Blt(NeoBufferContext<T_COLOR_FEATURE>, uint16_t, int16_t, int16_t, int16_t) [with T_COLOR_FEATURE = NeoGrbFeature; T_FILE_METHOD = fs::File; uint16_t = short unsigned int; int16_t = short int]'
```

Makuna, 2016-09-27:
The read will be fixed in my code.  The structures should be cast.  While it is still true they read should also be just as compatible (void*). 
```
size_t result;

        result = _file.read((uint8_t*)&bmpHeader, sizeof(bmpHeader));
```

sfranzyshen, 2016-09-27:
Thank you for you response to this issue! So, from what I gathered ... I edit my FS.h to change the SeekMode to have a default value of SeekSet and edit the (your) NeoBitmapFile.h to cast to (uint8_t*)? and things should have worked - but

```
Build options changed, rebuilding all
In file included from /home/user/Arduino/libraries/NeoPixelBus_by_Makuna/src/NeoPixelBus.h:50:0,
                 from /home/user/Arduino/NeoPixelBitmap/NeoPixelBitmap.ino:12:
/home/user/Arduino/libraries/NeoPixelBus_by_Makuna/src/internal/NeoBitmapFile.h: In instantiation of 'bool NeoBitmapFile<T_COLOR_FEATURE, T_FILE_METHOD>::Begin(T_FILE_METHOD) [with T_COLOR_FEATURE = NeoGrbFeature; T_FILE_METHOD = fs::File]':
/home/user/Arduino/NeoPixelBitmap/NeoPixelBitmap.ino:80:32:   required from here
/home/user/Arduino/libraries/NeoPixelBus_by_Makuna/src/internal/NeoBitmapFile.h:116:16: error: no matching function for call to 'fs::File::read(BitmapInfoHeader*, unsigned int)'
         result = _file.read(&bmpInfoHeader, sizeof(bmpInfoHeader));
                ^
/home/user/Arduino/libraries/NeoPixelBus_by_Makuna/src/internal/NeoBitmapFile.h:116:16: note: candidates are:
In file included from /home/user/Arduino/NeoPixelBitmap/NeoPixelBitmap.ino:16:0:
/home/user/.arduino15/packages/esp8266/hardware/esp8266/2.3.0/cores/esp8266/FS.h:59:9: note: virtual int fs::File::read()
     int read() override;
         ^
/home/user/.arduino15/packages/esp8266/hardware/esp8266/2.3.0/cores/esp8266/FS.h:59:9: note:   candidate expects 0 arguments, 2 provided
/home/user/.arduino15/packages/esp8266/hardware/esp8266/2.3.0/cores/esp8266/FS.h:65:12: note: size_t fs::File::read(uint8_t*, size_t)
     size_t read(uint8_t* buf, size_t size);
            ^
/home/user/.arduino15/packages/esp8266/hardware/esp8266/2.3.0/cores/esp8266/FS.h:65:12: note:   no known conversion for argument 1 from 'BitmapInfoHeader*' to 'uint8_t* {aka unsigned char*}'
exit status 1
Error compiling for board Generic ESP8266 Module.
```

Makuna, 2016-09-27:
I only included one location as an example.  Every read that passes a structure will need to be cast in the same way.

sfranzyshen, 2016-09-28:
Got It! Thank You!

Makuna, 2016-12-19:
with the latest version of my library and the GIT version of Esp8266/Arduino it should work. 

With the released version of Esp8266/Arduino, you will have to modify the fs.h file as you did before.

I am leaving this open until the official Esp Arduino release contains the fixes.

Torxgewinde, 2017-03-17:
In addition I had to remove the "const" keyword as follows to use the GetPixelColor function:
typename T_COLOR_FEATURE::ColorObject GetPixelColor(int16_t x, int16_t y)
otherwise the compilation failed with a message similar "to discards qualifiers [-fpermissive]".

Makuna, 2017-03-17:
@Torxgewinde Was this with the Arduino IDE?  What warning levels do you have?

Torxgewinde, 2017-03-18:
Hi,
Yes, Arduino IDE 1.6.12 with stable Arduino ESP8266 compiler (+the minor but crucial modification in FS.h). Warning levels are set to "standard".

The error as copied out of the Arduino IDE:
```
Arduino: 1.6.12 (Linux), Board: "NodeMCU 1.0 (ESP-12E Module), 80 MHz, 230400, 4M (3M SPIFFS)"

In file included from /home/tom/Sketches/libraries/NeoPixelBus_by_Makuna/src/NeoPixelBus.h:51:0,
                 from /home/tom/Sketches/Ringlicht/a40_LED.ino:22:
/home/tom/Sketches/libraries/NeoPixelBus_by_Makuna/src/internal/NeoBitmapFile.h: In instantiation of typename T_COLOR_FEATURE::ColorObject NeoBitmapFile<T_COLOR_FEATURE, T_FILE_METHOD>::GetPixelColor(int16_t, int16_t) const [with T_COLOR_FEATURE = NeoGrbFeature; T_FILE_METHOD = fs::File; typename T_COLOR_FEATURE::ColorObject = RgbColor; int16_t = short int]:

/home/tom/Sketches/Ringlicht/a40_LED.ino:97:59:   required from here
/home/tom/Sketches/libraries/NeoPixelBus_by_Makuna/src/internal/NeoBitmapFile.h:185:45: error: passing const NeoBitmapFile<NeoGrbFeature, fs::File> as this argument of bool NeoBitmapFile<T_COLOR_FEATURE, T_FILE_METHOD>::readPixel(RgbColor*) [with T_COLOR_FEATURE = NeoGrbFeature; T_FILE_METHOD = fs::File] discards qualifiers [-fpermissive]
         if (!seek(x, y) || !readPixel(&color))
                                             ^
/home/tom/Sketches/libraries/NeoPixelBus_by_Makuna/src/internal/NeoBitmapFile.h:185:23: error: passing const NeoBitmapFile<NeoGrbFeature, fs::File> as this argument of bool NeoBitmapFile<T_COLOR_FEATURE, T_FILE_METHOD>::seek(int16_t, int16_t) [with T_COLOR_FEATURE = NeoGrbFeature; T_FILE_METHOD = fs::File; int16_t = short int] discards qualifiers [-fpermissive]
         if (!seek(x, y) || !readPixel(&color))
                       ^
exit status 1
Fehler beim Kompilieren für das Board NodeMCU 1.0 (ESP-12E Module).

Dieser Bericht wäre detaillierter, wenn die Option
Ausführliche Ausgabe während der Kompilierung
in Datei -> Voreinstellungen aktiviert wäre.
```

The relevant *.ino files is attached.
[a40_LED.ino.txt](https://github.com/Makuna/NeoPixelBus/files/852278/a40_LED.ino.txt)

And as soon as I figured out to remove "const" it compiles fine and the resulting firmware works as expected. Strange, isn't it?

Regards,
Tom

Makuna, 2018-08-17:
This is due to the FS.read not being const, as it modifies the internal read position as it reads, thus GetPixelColor can't be const.

https://github.com/Makuna/NeoPixelBus/pull/223

Makuna, 2018-08-23:
fixed

closed 2018-08-23 by Makuna