Makuna/NeoPixelBus #138: Add a filter class to modify the color/brightness

protolyser, 2016-11-11:
'NeoPixelBus <NeoRgbFeature, Neo800KbpsMethod, MyBrightnessFilter> strip()'

ctrowat, 2016-11-17:
If you are trying to judge whether people are interested in this feature I'd like say - I am also interested, and thanks so much for all your hard work on this project, it is amazing!

Makuna, 2016-11-17:
@ctrowat I wish there was a vote feature (or maybe there is and I am not able to do so).  But in the absence of a vote feature; please feel free to "me too" or "+1" a comment to any issue.  

It does help prioritize features beyond the ones I personally want.

protolyser, 2016-11-17:
+1 is the usual thing to do I guess (pretty common @fontawesome icon requests.
@ctrowat depending on what you try to do you can have a look at how I'm doing the brightness stuff as a workaround (till the new filter class is implemented) on gitter https://gitter.im/Makuna/NeoPixelBus?utm_source=badge&utm_medium=badge&utm_campaign=pr-badge

Makuna, 2016-12-06:
Could I get some feedback on an approach?

Here is the summary on the technical details of two approaches to get global dimming (or any color modification).  Both come with tradeoffs.

1)  Modify on SetPixelColor():
If dimming is implemented by modifying a color when you set it; then when you retrieve said color it will be different.  This is due to fact that the original color is multiplied/divided by the dim value when stored, and reversed when retrieved.  
The math causes the loss of information.  Consider the Red value 128 dimmed to 10%.  The value 128 is multiplied by 0.1 to get the final value of 13 (or 12 depending on truncating or rounding as the colors are stored as byte values).  To retrieve this value, the 13 would be multiplied by 10 and you get 130; not the value you originally set.   Consider 193 dimmed by 75% (144) would be retrieved as 192.
This is how Adafruit and others treat DIM support and the reason I don't want to implement it this way.  
But it has some efficiencies.  Only one buffer is needed, and no extra work is done when the actual buffer is sent out the wire.
I didn't include this originally as it is easy for the user to accomplish this by modifying the color before calling SetPixelColor using a stored global value.

2) Modify on Show():
If dimming is implemented by creating a second buffer (or using an existing one already required due to hardware requirements of the Method); then SetPixelColor and GetPixelColor will always match.  
But this will incur extra work when you call `Show()` that the pixel values be modified by the global dim value before being placed into the second buffer; and this second buffer is the one sent out the wire.
Further, not all methods have an extra  buffer, so in some cases the memory footprint will be doubled to provide this secondary buffer. The Esp8266  two platform methods DMA and AsyncUart are the only ones with a secondary buffer due to hardware/method requirements; so all others will require another buffer.

ctrowat, 2016-12-06:
I'm happy with either one - in my projects I have no need to read pixel data back out and I use the DMA mode on an ESP8266 so I already have the second buffer.  I was working on adding global dimming to my project and when I planned it out my approach was most similar to your method number 1 because in some of my animations I don't draw every pixel so I figured it had less overhead.

Makuna, 2016-12-09:
I really need to know what you expect from this.  Playing with Adafruits setBrightness(), I find it does weird things that I would not expect.  calling it repeatedly with incrementally changing values (animating the whole strip to dim and back up); while it dims fine; as it brightens back up it starts to dim again (all due to the horrible assumptions you have to make with the math).  
But if you expect to just call it once; it works fine.
Sigh, this is getting back to my original design of leaving it off and letting the user implement the filter in their code so they understand the tradeoffs.
Quote from Adafruits code: 

> If there's a significant step up in brightness, the limited number of steps (quantization) in the old data will be quite visible in the re-scaled version.  For a non-destructive change, you'll need to re-render the full strip data.  C'est la vie.

SO, this could never be used to animate dim and bright of the entire strip; it could be only used to set a "level" that you want when you apply color changes.

Makuna, 2016-12-11:
Ok, I have added brightness support; it could be used to provide other filtering but I could not come to a good method that didn't make the use of the library without far more complex.

See the example NeoPixelBrightness and the use of the new NeoPixelBrightnessBus object.

dotorg, 2017-01-23:
So, a bit of feedback on this ...

I think the NeoPixelBrightnessBus would be "safer" if it either applied the brightness when the code that is sending data to the strip read its, or maintains a separate buffer for it internally, so calls to set a color and calls to get a color always agree on the color of a pixel. The way it works now means a lot of clunky work-around for anything but the simplest of use-cases. Just as a couple examples, if you set a pattern into a strip, adjust the brightness, then want to save the current pattern (say, for a reboot or something), the saved pattern will have the brightness applied already. When you go to reload the pattern, there's no way to get it back in there *and* get the brightness right, because you'll either adjust the values when you set them or when you set the old brightness back. It also means if you want to generically handle dimming and brightening, you're left having to keep an external array of colors. 

The code to do that isn't particularly complicated, it just seems, at first glance, that the vast majority of use cases are going to need to do that, so the framework really should handle it internally. Then the brightness setting would be really a brightness of the strip... and could be changed any time you want. 

On a semi-related note, the brightness probably should be in percent (0-100) and either apply, or at least support applying, the gamma adjustment, so 50% is really 50%. The current implementation makes it necessary to calculate all of that externally, as well. 

I worked around the limitation currently by just keeping a separate copy of the colors and reloading them any time the brightness is changed, and scaling code to map the brightness against the gamma table... but I've been considering going into the core library and adding an Rfilter, Gfilter and Bfilter float and applying them as the data is fed out to the device itself. So I'd be able to set them all to 0.5f and get an un-corrected 50%, or I could warm up the strip by running 1f for everything but blue and 0.95f, just as an example.  I decided not to for now because a) I don't like forking libraries, and b) the C code doesn't know about the differences between GBR, RGB, RGBW, etc... so it would've taken some refactoring.

Makuna, 2017-01-23:
It is provided to be a "compatibility" layer with the Adafruit brightness features.  It currently works exactly like theirs.  I don't like it, but there has been a lot of demand for it.

I am still trying to come up with a solution to provide filtering without making it harder to use for novice consumers.  
The secondary buffer issue will be required for some methods; but there are some methods that already have one (on Esp8266 DMA method) that could be also used for it making it a more complex subject to solve quickly.

Makuna, 2017-01-24:
https://github.com/Makuna/NeoPixelBus/pull/158

dotorg, 2017-01-25:
I haven't built a library from source before, I need to read up on how to do that, and will give it a try. 

Looking at the code, though, I think that's a *great* way to do it. It componentizes things in a cleaner way, and allows dimming (or, any other actions like tinting -- like f.lux -- without having to reset things over and over again. My current dimming code on a ramp up or down of the strip has to reset the colors every couple of steps so things stay colored right, for example...

I probably won't have time to experiment with it before tomorrow, but I will definitely do it then.

Makuna, 2017-01-25:
No need to build ;-)
Look at the front page of this Repository, the readme file that is displayed lower has the instructions; its actually pretty simple; but the repo will sit in the same location as the Library Manager installed library, so make sure to manually delete ALL the files present in the destination and when you want to switch back to the official library, delete them all again and use the library manager to get them back.

dotorg, 2017-01-26:
I started making the changes ... its actually more code than I expected to touch. I understand why its structured the way it is, but its a little convoluted to retrofit into existing code. You basically have to touch everything that had been previously touching the pixels, and update the Dib, then render it while applying the shader. It would've been "cleaner" to do if the shader was registered with the NeoBufferContext directly. 

The one problem I've run into, though, because of it -- without a way to know the DIB is "dirty", you basically have to code all over the place to re-render the DIB into the bus itself. That's slow as molasses if you're doing it in an animator, and you're running an animator per-pixel, because every one of them running has to re-render the entire set to the bus. 

What I did temporarily is add a bool in my main sketch file "rerenderDib" that anything that touches the "shader" settings or changes anything in the DIB sets to true, and at the end of my loop before calling Show on the bus object, if thats set, it re-renders things.  Its easy enough, but it may make sense to track that in the Dib itself. (It'd need to be something that can be set, too -- because anything changing the shader also will need to trigger a re-render.)

Generally, though, I like it. Its better than the Brightness bus.

The only thing I'm left trying to figure out (and, admittedly, its because my C++ skills are a solid 25 years out of date) is how to set and reference the shader dynamically. I can create a single monolithic shader for a bunch of different situations and make them one big stack of if statements, but it'd be easier to just swap them out.

Makuna, 2017-01-26:
I will add a Dirty flag inside the dib so it will only render if anything has touched it.  Similar in the way the bus works.  Generally you want to call render right before the show, and should avoid having to many places you call it.  I suggest once in your main loop.

There is no way to swap the shader; you will either have to embed the switch in the "one" shader (the apply has a switch statement), or at the location you call Render have a switch statement (my recommendation).

dotorg, 2017-01-26:
> I will add a Dirty flag inside the dib so it will only render if anything has touched it. 

That'll help... but make sure there's an "MarkDirty()" call on it, or something like that. Otherwise it wouldn't know that, for example, the brightness setting in your example shader had changed. Or, add a Dirty flag to the shader definition, and notify the shader class when the rendering has completed, so the shader can mark itself dirty and know when to reset that flag. (That may be a better option, because you can change things willy-nilly and not have to remember to do it manually as a consumer, plus it'll keep things straight if you've got multiple DIB instances and want to share a shader instance.)

Makuna, 2017-01-26:
https://github.com/Makuna/NeoPixelBus/pull/159

dotorg, 2017-01-28:
I pulled the updates this morning and removed my dirty flag code. Works like a charm!

IMO, this is a far better solution to the original problem. Its both easier to reliably implement, and far more powerful. 

Thanks!

Makuna, 2017-01-28:
I will be expanding this to the other image classes (that are color feature specific); which will provide performance benefits but will make it more complex/specific to how to "effect" the pixels as they will be in their device format rather than RgbColor format.

closed 2018-08-16 by Makuna