Makuna/NeoPixelBus #139: ESP-01 not working?

j4k3, 2016-11-14:
Sorry for asking a beginners question here, I dont know where else to ask don't have a clue what else to try. Maybe someone can guide me in the right direction.

I've had no problems driving a WS2812 strip running the DMA variant of this library on a Sparkfun Thing dev board. I've got a bunch of ESP-01, but I have no luck with these. I read the documentation a bunch of times, I understand that I have got to use the UART method on these. So, I worte up a small sketch that I expect to make an LED blink and the WS2812 to cycle through colors. Well, the LED blinks, so I know the code was uploaded correctly, but the strip stays off. I know the WS2812 are working, and the same code is working on the dev board. What am I doing wrong? I'm not sure if I understood the instructions on what pin to use. The documentation says "Due to using the TXD1/GPIO2 pin", but GPIO2 and TXD1 are seperate ones, aren't they? Does that mean, I can use either one, or what am I not getting about this? What pin numbers are these assigned to in Arduino? I tried quite a few numbers and pins, but it seems I'm still missong something.

penfold42, 2016-11-14:
A good table of gpio and alternate functions is at 
http://www.esp8266.com/wiki/lib/exe/detail.php?id=esp8266_gpio_pin_allocations&media=pin_functions.png

Gpio2 and txd1 (not txd0) are the same pin.

Contrary to the documentation DNA mode does work on the esp01 as gpio3 is rxd0 and available

j4k3, 2016-11-14:
Thanks for the reply. Going to try as soon as I can.

Makuna, 2016-11-14:
J3k4, this is a good example to ask on gitter.  The link is both in the readme and the wiki first pages.  [image]

penfold42, feel free to update the docs; its just a wiki and anyone can edit it (as I can always revert the edit if I don't like the changes but I have never needed to so far).

j4k3, 2016-11-14:
I've completely skipped that part, you're absolutely right, of course. I'll ask over there if I have any further questions. I'll just close this issue. Thanks.

closed 2016-11-14 by j4k3