Makuna/NeoPixelBus #140: chipKit support

GrahamM, 2016-11-18:
I note that the NeoPixelBus library doesn't currently support chipKit PIC32 based systems. I spent time creating suitable functions to support bitbanging WS2812 devices on chipKit boards. In theory the Dot Star stuff should work, but I don't have anything to test it with.

Attached is a diff with the changes if you're interested in integrating them.
[NeoPixelBusPatch.txt](https://github.com/Makuna/NeoPixelBus/files/599858/NeoPixelBusPatch.txt)

Makuna, 2016-11-18:
I don't have one to test with so it would be hard for to integrate it with confidence.
I created this Git project so that you can actually place the "enlistment" (clone) into place in the doc/Arduino/libraries/ folder; making it easy to development/modify/ and share changes.  
You could do this and then just create a pull which I could review.

GrahamM, 2016-11-19:
Sure, I'll give it a go.

closed 2016-11-19 by GrahamM

Makuna, 2016-11-19:
If you have any questions as you do this; you can ask on the gitter chat associated with this project (see Readme)