Makuna/NeoPixelBus #144: ESP8266 halts when more then 240 pixels are configured

jan-gerard, 2016-12-12:
The ESP8266-12 halts when more then 240 pixels are configured. I can not configure more, since then the program does not start at all. Is this normal? Could this be increased? What can I do to troubleshoot, in case this should work normally?

Makuna, 2016-12-12:
Please provide a minimal sketch (no more code than is needed) that demonstrates the problem.

This is not normal as long as you have enough ram.  See [NeoEsp8266Dma800KbpsMethod](https://github.com/Makuna/NeoPixelBus/wiki/NeoPixelBus-object#neoesp8266dma800kbpsmethod) for more details on memory use as it is the default method on Esp8266 platform.

jan-gerard, 2016-12-12:
Hi, I don't know what I did wrong, but now I simplified the sketch to to upload here, I got it working. So probably there wasn't a neopixel library issue, Thanks for your reply anyway.

closed 2016-12-12 by jan-gerard