Makuna/NeoPixelBus #150: Support for circular displays?

jteiwes, 2017-01-06:
The Matrix Panel Support can be used to structure a rectangular or even quadratic display of ws2812 LEDs. 

Currently i am trying to apply this to a circular display. I made a sketch of the layout here:

[image]

As far as i understand it this display cannot be structured using the matrix panel support since every line has a different number of columns. 

Can anybody give a hint on where to integrate this into the library or how to achieve this?

Makuna, 2017-01-08:
A beginning of documenting this as a Wiki topic.
https://github.com/Makuna/NeoPixelBus/wiki/Spiral-Topography

Makuna, 2017-01-14:
#154

jteiwes, 2017-01-15:
Thanks a lot, i will have a try at this one.. Looks really good!

closed 2017-01-24 by Makuna